MikePopoloski/slang #1186: Binding not finding

ChangyangZen, 2024-12-06:
**Describe the bug**
When I use Slang to parse the Ariane MMU SystemVerilog assertion's property module, Slang can not recognize some signals from TLB even if I put the binding file and tlb.sv in the command line.

The command line output is as following shows:
 slang  -Wimplicit-port-type-mismatch -y 'ariane/include/' -y 'ariane/src/riscv-dbg/src' ./ariane/include/riscv_pkg.sv ./ariane/src/riscv-dbg/src/dm_pkg.sv ./ariane/include/ariane_pkg.sv  ./ariane/src/tlb.sv ./ariane/src/ptw.sv ariane/src/mmu.sv mmu_prop.sv mmu_bind.sv
Top level design units:
    mmu

ariane/src/mmu.sv:185:10: warning: implicit named port 'pmpaddr_i' of type 'logic[15:0][53:0]' connects to value of inequivalent type 'logic[15:0][55:0]' [-Wimplicit-port-type-mismatch]
        .pmpaddr_i,
         ^~~~~~~~~
SVA/mmu/sva/mmu_prop.sv.2.save:74:65: error: could not resolve hierarchical path name 'i_itlb'
asgpt__flush_i_tlb_flush: assert property (flush_i |=> (mmu_prop.i_itlb.flushed && mmu_prop.i_dtlb.flushed));
                                                                ^~~~~~~
SVA/mmu/sva/mmu_prop.sv.2.save:74:92: error: could not resolve hierarchical path name 'i_dtlb'
asgpt__flush_i_tlb_flush: assert property (flush_i |=> (mmu_prop.i_itlb.flushed && mmu_prop.i_dtlb.flushed));
                                                                                           ^~~~~~~
**To Reproduce**
1. clone the Ariane repo: 
git clone https://github.com/morenes/cva6.git ariane
cd ariane
git checkout autosva
git submodule update --init --recursive --force src
cd ..
2. Edit mmu_prop.sv and mmu_bind.sv
vi mmu_prop.sv
`module mmu_prop 
  import ariane_pkg::*; #(
    parameter INSTR_TLB_ENTRIES     = 4,
    parameter DATA_TLB_ENTRIES      = 4,
    parameter ASID_WIDTH            = 1,
    parameter ariane_pkg::ariane_cfg_t ArianeCfg = 0//ariane_pkg::ArianeDefaultConfig
) (
    input  logic                            clk_i,
    input  logic                            rst_ni,
    input  logic                            flush_i,
    input  logic                            enable_translation_i,
    input  logic                            en_ld_st_translation_i,   // enable virtual memory translation for load/sto>

    /*AUTOSVA
    itlb_lookup: itlb_req -IN> itlb_res
    itlb_req_val = icache_areq_i.fetch_req
    itlb_req_rdy = icache_areq_o.fetch_valid
    [riscv::VLEN+ASID_WIDTH-1:0] itlb_req_stable = {asid_i,icache_areq_i.fetch_vaddr}
    itlb_res_val = icache_areq_o.fetch_valid

    dtlb_lookup: dtlb_req -IN> dtlb_res
    dtlb_req_val = lsu_req_i
    [riscv::VLEN+ASID_WIDTH-1:0] dtlb_req_stable = {asid_i, lsu_vaddr_i};
    dtlb_res_val = lsu_valid_o
    */
    // IF interface
    input  icache_areq_o_t                  icache_areq_i,
    input icache_areq_i_t                  icache_areq_o,

    // LSU interface
    // this is a more minimalistic interface because the actual addressing logic is handled
    // in the LSU as we distinguish load and stores, what we do here is simple address translation
    input  exception_t                      misaligned_ex_i,
    input  logic                            lsu_req_i,        // request address translation
    input  logic [riscv::VLEN-1:0]          lsu_vaddr_i,      // virtual address in
    input  logic                            lsu_is_store_i,   // the translation is requested by a store
    // if we need to walk the page table we can't grant in the same cycle
    // Cycle 0
    input logic                            lsu_dtlb_hit_o,   // sent in the same cycle as the request if translation hi>    input logic                            lsu_valid_o,      // translation is valid
    input logic [riscv::PLEN-1:0]          lsu_paddr_o,      // translated address
    input exception_t                      lsu_exception_o,  // address translation threw an exception
    // General control signals
    input riscv::priv_lvl_t                 priv_lvl_i,
    input riscv::priv_lvl_t                 ld_st_priv_lvl_i,
    input logic                             sum_i,
    input logic                             mxr_i,
    // input logic flag_mprv_i,
    input logic [riscv::PPNW-1:0]           satp_ppn_i,
    input logic [ASID_WIDTH-1:0]            asid_i,
    input logic [ASID_WIDTH-1:0]            asid_to_be_flushed_i,
    input logic [riscv::VLEN-1:0]           vaddr_to_be_flushed_i,
    input logic                             flush_tlb_i,
    // Performance counters
    input logic                            itlb_miss_o,
    input logic                            dtlb_miss_o,
    // PTW memory interface
    input  dcache_req_o_t                   req_port_i,
    input dcache_req_i_t                   req_port_o,
    // PMP
    input  riscv::pmpcfg_t [15:0]           pmpcfg_i,
    input  logic [15:0][riscv::PLEN-1:0]    pmpaddr_i
);

// Immediate assertion: Check that flush_i is a 1-bit signal
// This ensures that the signal is correctly defined as a 1-bit input.
asgpt__flush_i_bitwidth: assert property (flush_i inside {1'b0, 1'b1});

// Concurrent assertion: Check that when flush_i is asserted, flush_tlb_i is also asserted in the same cycle
// This ensures that the flush request is properly propagated to the TLB modules.
asgpt__flush_i_propagation: assert property (flush_i |-> flush_tlb_i);

// Concurrent assertion: Check that when flush_i is asserted, both i_itlb and i_dtlb are flushed in the next cycle
// This ensures that the TLBs are flushed as expected when flush_i is asserted.
asgpt__flush_i_tlb_flush: assert property (flush_i |=> (mmu_prop.i_itlb.flushed && mmu_prop.i_dtlb.flushed));`

Edit mmu_bind.sv
`bind mmu mmu_prop
  #() u_mmu_sva(.*);
`
3. run slang command
slang  -Wimplicit-port-type-mismatch -y 'ariane/include/' -y 'ariane/src/riscv-dbg/src' ./ariane/include/riscv_pkg.sv ./ariane/src/riscv-dbg/src/dm_pkg.sv ./ariane/include/ariane_pkg.sv  ./ariane/src/tlb.sv ./ariane/src/ptw.sv ariane/src/mmu.sv mmu_prop.sv mmu_bind.sv

**Additional context**

MikePopoloski, 2024-12-06:
Where are `i_itlb` and `i_dtlb` declared? They don't appear to be members of mmu_prop, so it's not surprising that they can't be found. Are you missing some other code some where?

ChangyangZen, 2024-12-06:
Hi Mike, `i_itlb` is defined in mmu.sv as following:
`   tlb #(
        .TLB_ENTRIES      ( INSTR_TLB_ENTRIES          ),
        .ASID_WIDTH       ( ASID_WIDTH                 )
    ) i_itlb (
        .clk_i            ( clk_i                      ),
        .rst_ni           ( rst_ni                     ),
        .flush_i          ( flush_tlb_i                ),

        .update_i         ( update_ptw_itlb            ),

        .lu_access_i      ( itlb_lu_access             ),
        .lu_asid_i        ( asid_i                     ),
        .asid_to_be_flushed_i  ( asid_to_be_flushed_i  ),
        .vaddr_to_be_flushed_i ( vaddr_to_be_flushed_i ),
        .lu_vaddr_i       ( icache_areq_i.fetch_vaddr  ),
        .lu_content_o     ( itlb_content               ),

        .lu_is_2M_o       ( itlb_is_2M                 ),
        .lu_is_1G_o       ( itlb_is_1G                 ),
        .lu_hit_o         ( itlb_lu_hit                )
    );
`
So is `i_dtlb`: 
`    tlb #(
        .TLB_ENTRIES     ( DATA_TLB_ENTRIES             ),
        .ASID_WIDTH      ( ASID_WIDTH                   )
    ) i_dtlb (
        .clk_i            ( clk_i                       ),
        .rst_ni           ( rst_ni                      ),
        .flush_i          ( flush_tlb_i                 ),

        .update_i         ( update_ptw_dtlb             ),

        .lu_access_i      ( dtlb_lu_access              ),
        .lu_asid_i        ( asid_i                      ),
	      .asid_to_be_flushed_i  ( asid_to_be_flushed_i   ),
	      .vaddr_to_be_flushed_i ( vaddr_to_be_flushed_i  ),
        .lu_vaddr_i       ( lsu_vaddr_i                 ),
        .lu_content_o     ( dtlb_content                ),

        .lu_is_2M_o       ( dtlb_is_2M                  ),
        .lu_is_1G_o       ( dtlb_is_1G                  ),
        .lu_hit_o         ( dtlb_lu_hit                 )
    );`
I do have bind mmu with mmu_prop in the command line options, also put tlb.sv there.

MikePopoloski, 2024-12-06:
If `i_tlb` is in `mmu` then why are you trying to access it via `mmu_prop` which is a different module / instance? That will not work.

ChangyangZen, 2024-12-06:
I was using this signal for the assertion in the mmu_prop.sv, and I bind mmu with mmu_prop.

MikePopoloski, 2024-12-06:
Does this work in any other tool? I don't think you are using bind directives correctly.

ChangyangZen, 2024-12-06:
Yes it works in JasperGold.

MikePopoloski, 2024-12-06:
Perhaps you're feeding a slightly different design to JasperGold?

From what I can tell, after your bind is applied this what the relevant hierarchy looks like:
```
mmu
 |--> i_tlb
 |--> mmu_prop
      |--> asgpt__flush_i_tlb_flush (assertion)
             |--> mmu_prop.i_itlb.flushed (hierarchical reference)
```
The `mmu_prop` prefix on your hierarchical reference ensures that it finds the `mmu_prop` module and not the `mmu` module above it that contains `i_tlb`

ChangyangZen, 2024-12-06:
I fed the same design to Jasper. But I think you are right about the hierarchy.
It turns out that Jasper complains nothing with mmu_prop.i_itlb. Even with undefined identifier like `potato.flush_i` (`flush_i` is a valid signal), Jasper only sends a warning as the following:
`WARNING (WNL021): */mmu_prop.sv(72): external reference "potato.flush_i" remains unresolved and will be treated as a free net.`

MikePopoloski, 2024-12-09:
Ok, it sounds like there's nothing to do here for slang then?

ChangyangZen, 2024-12-09:
I think so, you can close this issue, thank you for answering me!

closed 2024-12-09 by MikePopoloski