MikePopoloski/slang #1193: [bug][slang][parser]

likeamahoney, 2024-12-24:
Hi, all!

`slang` does not emit an error for this case with nested attributes:

```verilog
    (* \%id_1  , \%id_9 = - (* \%id_13   *) null *) module  Interf24;
   endmodule ;
```

The standard prohibits the use of nested attributes (according to 5.12 LRM section):

> Nesting of attribute instances is disallowed. It shall be illegal to specify the value of an attribute with a
> constant expression that contains an attribute instance.

Does `slang` doing it right or is this a bug?

MikePopoloski, 2024-12-24:
Yes, this looks like a bug.

likeamahoney, 2024-12-24:
> Yes, this looks like a bug.

Thank you for confirming the bug.

I suggested my fix here - https://github.com/MikePopoloski/slang/pull/1194

likeamahoney, 2024-12-24:
Also another test like this, with the fix above, has clearer error messages, but not just "expected endmodule":
```verilog
extern   (* \%id_1  , \%id_9 = - (* \%id_13   *) null *) module mod;
```

closed 2024-12-25 by MikePopoloski