MikePopoloski/slang #1195: Genvar declared in loops does not seems to be recorded as symbols

suzizecat, 2024-12-25:
**Describe the bug**
When declaring the iterator of a `for-generate` loop as a genvar directly in the `for` statement itself, the iterator variable does not seems to show up as a symbol within the AST. 

**To Reproduce**

Using the following SV:
```systemverilog
module myMod();
	generate
		for(genvar i = 0; i<3; i++) begin : g_test
		end
	endgenerate;
endmodule;
```
a custom made "AST Walker" (`sv-explorer-exe` target from https://github.com/suzizecat/slang-lsp-tools ) reports the following elements:

```text
 # Scope  of kind CompilationUnit with name  spanning in samples/genvar.sv:1:1:6:11
   - Symbol of kind EmptyMember with name   without syntax node.
 - Symbol of kind Instance with name myMod  of type myMod samples/genvar.sv:1:1:6:10
   # Scope  of kind GenerateBlockArray with name g_test spanning in samples/genvar.sv:3:3:4:6
     # Scope  of kind GenerateBlock with name  spanning in samples/genvar.sv:3:31:4:6
       - Symbol of kind Parameter with name i  without syntax node.
     # Scope  of kind GenerateBlock with name  spanning in samples/genvar.sv:3:31:4:6
       - Symbol of kind Parameter with name i  without syntax node.
     # Scope  of kind GenerateBlock with name  spanning in samples/genvar.sv:3:31:4:6
       - Symbol of kind Parameter with name i  without syntax node.
   - Symbol of kind EmptyMember with name   without syntax node.
```

We can see that `i` is recorded as a parameter of generate blocks which I can understand. However:
 1. It does not have any attached syntax node
 2. I would have expected a `GenvarSymbol` alongside within the `GenerateBlockArray` (which is the scope covering the `for` statement) 

**Additional context**

Using a standard ASTVisitor handling `GenvarSymbol` does not yield any result in this situation.

This have been tested using current master (6f9e7bfa606b3a0d2a9258e1baf46f33086a8188) as well as the 7.0 tag.

MikePopoloski, 2025-01-06:
When the genvar is declared inline to the loop like this there isn't really a symbol created, slang just performs the generate loop and creates each iteration's block. Since there aren't any observable expressions nothing would reference it anyway, so I'm not sure what you would do with it. I guess it wouldn't be too hard to fabricate one and put it in the GenerateBlockArray though if really needed.

The implicit parameters don't have a syntax node because they are sort of implicitly generated (as described in the LRM). I suppose they could take on the syntax of the genvar if it's declared inline but it's kind of weird if the genvar is not inline.

Can you describe what you're trying to do and what you'd like to see happen?

suzizecat, 2025-01-07:
So, in the exact code extract that I put, there is indeed no reference. This was a MWE, used to reduce the AST to the strict minimum.
However, this code is often (always ?) used with `i` referenced somewhere. In example, my actual test file (still a silly MWE) is the following (from my [slang-lsp-tools repo](https://github.com/suzizecat/slang-lsp-tools/blob/master/samples/designs/sub.sv) ):
```systemverilog
module sub #(
    parameter int K_NOUT = 1
)  (
    input logic i_clk,
    input logic i_rst_n,
    output logic [K_NOUT-1:0] o_out
);

    logic [K_NOUT-1:0] toggle;

    assign o_out = toggle;

    generate
        genvar t;
        for(genvar i = 0; i < K_NOUT; i++) begin : g_loop
            logic test;
            always_ff @(posedge i_clk or negedge i_rst_n) begin : p_ff_process
                if(~i_rst_n) begin
                    toggle[i] <= 0;
                end else begin
                    toggle[i] <= ~toggle[i];
                end
            end
        end
    endgenerate
endmodule 
```
For reference, in this example the `t` variable is in the AST as I would expect.

> Can you describe what you're trying to do and what you'd like to see happen?

I would expect the `i` symbol to be present in the AST, as any other signal/variable/parameter/... (used or not) would be.
And I would expect it to be in the scope starting at the `for` keyword, so the reference within the for statement itself can be found as well.

MikePopoloski, 2025-01-08:
Ok, this is simple enough to do so I'll just do it. I'm still curious what you want to use it for; genvars are basically syntactical noise that don't really do anything. Probably the language should just deprecate of them and force generate loops to always just declare their iteration variables inline.

suzizecat, 2025-01-08:
> Ok, this is simple enough to do so I'll just do it.

Thanks a lot !

> I'm still curious what you want to use it for; genvars are basically syntactical noise that don't really do anything

Oh, for that point it is simple, on my language server, I just want that when you select any valid "symbol" (in a broad sense, that would be roughly any user-defined identifier, including typename) in the source code, you can lookup the definition and the references of this symbol.

This also allows me to perform renaming and scope-aware completion (admittedly, completing `i` is not very useful). I just raised the point because I stumbled on it when testing my indexer.

When looking up references, I am detecting them using the CST, and a very broad detection (as of today, I catch all `syntax::IdentifierNameSyntax` and `syntax::IdentifierSelectNameSyntax`) which correctly detects the `i` usage while the symbol definition has not been found.

MikePopoloski, 2025-01-12:
Added in 06b1a766a3c329ecf63f55cf3a0b6fd91a8c555d