MikePopoloski/slang #1196: Build fail on Ubuntu 24.04 LTS

hpretl, 2025-01-02:
**Describe the bug**
I am building Slang in Ubuntu 24.04. With some of the latest changes, a build crash has been introduced.

**To Reproduce**
This works fine: 6f9e7bfa606b3a0d2a9258e1baf46f33086a8188
This causes a crash during build: f03ccdb83c873b9c618aa8db317b8003032d14d2

**Additional context**
Here is my build script:
```
git clone --filter=blob:none "${SLANG_REPO_URL}" "${SLANG_NAME}"
cd "${SLANG_NAME}" || exit 1
git checkout "${SLANG_REPO_COMMIT}"
cmake -B build -DSLANG_INCLUDE_TESTS=OFF
cmake --build build -j"$(nproc)"
cmake --install build --strip --prefix="${TOOLS}/${SLANG_NAME}"
```

MikePopoloski, 2025-01-02:
Can you explain what a build crash is? What output are you seeing? What compiler are you using?

hpretl, 2025-01-02:
I meant a build fail, sorry :-) I think it is fixed with dbfa01c48b86385f7d13f1786efd1eb56f939c73

closed 2025-01-03 by MikePopoloski