MikePopoloski/slang #1200: Finish case related warnings

MikePopoloski, 2025-01-04:
- Check completeness of small-sized bit vectors?
- Think about how priority, unique, unique0 apply
- Check applicability to all other forms of case statements
    - property exprs
    - pattern cases
    - randseq cases
    - case generates

MikePopoloski, 2025-11-20:
This is intentional. `unique` cases report an error in simulation if there's no match, so there's no reason to specify a default. `unique0` does not, so you may still wish to have a default case to handle that. This is just an optional warning, you don't have to enable it.

spomatasmd, 2025-11-21:
Fair, indeed it is an optional warning however the slang-server LSP enables it and that's how I stumbled onto it.
Still not 100% clear why would one pick `unique0` with a default, I was under the assumption that, from an inferred hardware point of view, `unique` and `unique0` are equivalent.

Need to dust off the document by C. Cummings since it's been a while from the last read and this periodically pops up every few months or so...

MikePopoloski, 2026-10-18:
Done in 2697e71bb23edfec88e375c2f30ed5d8e74f97e6

closed 2026-10-18 by MikePopoloski