MikePopoloski/slang #1207: [slang][bug] Mismatch between the description of the method and its body.

likeamahoney, 2025-01-10:
Hi @MikePopoloski 

[Here](https://github.com/MikePopoloski/slang/blob/dd7f9c0cd904c93cc6316aa2c6f6a17773a4e2be/include/slang/diagnostics/Diagnostics.h#L195) is the method description which tells that list of diagnostics would be filtered out according to provided list with diag codes (In other words, only diagnostics with matching diagnostic codes will be left).

But the method [implementation](https://github.com/MikePopoloski/slang/blob/dd7f9c0cd904c93cc6316aa2c6f6a17773a4e2be/source/diagnostics/Diagnostics.cpp#L168) does the opposite, it leaves only those diagnostics that are not present in the provided list. What was meant?

furkanusta, 2025-01-10:
I think the comment is correct

`Diagnostics that match any is filtered out` (This is what the comment says)
If you negate both sides of that comment it will be equivalent to this
`Diagnostics that match none is filtered in` (This is what the code does)

> the list of diagnostics would be filtered out according to provided list with diag codes (In other words, only diagnostics with matching diagnostic codes will be left).

Diagnostics with matching codes will be filtered out (meaning removed/skipped), only diagnostics with *non*-matching codes will be left

MikePopoloski, 2025-01-11:
Yes, the comment and the code seem correct to me. It removes any diagnostics from the set that match any of the given DiagCodes.

closed 2025-01-11 by MikePopoloski

likeamahoney, 2025-01-13:
Ok!  Thank you! I'm sorry for the misunderstanding