MikePopoloski/slang #1209: Bad sizing of expression inside a size cast

povik, 2025-01-11:
**Describe the bug**
On occurence of a sizing cast syntax like `(n)'(/expr/)` where `n` is a positive integer, the standard says the following:

> When changing the size, the cast shall return the value that a packed array type with a single [n-1:0] dimension would hold after being assigned the expression, where n is the cast size.

1800-2023, p. 141

As far as I can tell slang does not follow that, instead it might be applying self-determined sizing to `/expr/` irrespective the value of `n`.

**To Reproduce**

This test fails as of [f04d7a0](https://github.com/MikePopoloski/slang/commit/f04d7a0f10e9ae96ca9bdfb4de6dd6eb543f6ce9) (current master)

```
TEST_CASE("Size cast upstream influence") {
    ScriptSession session;
    auto value = session.eval("(3)'(1'b1 << 2)");
    CHECK(value.integer() == 4);
}
```

**Additional context**

On p. 140 the standard also says the below, but the mention of padding/truncation is somewhat vague and not what other tools seem to be following, exactly:

> If the casting type is a constant expression with a positive integral value, the expression in parentheses shall
be padded or truncated to the size specified. It shall be an error if the size specified is zero or negative.

Downstream issue povik/yosys-slang#78

MikePopoloski, 2025-01-12:
Indeed, it looks like I misinterpreted the rule. I agree the intent is that it behave the same as an implicit cast.

MikePopoloski, 2025-01-12:
Fixed in dfed278b003524c6aea02da34ea36fecd23e5feb

closed 2025-01-12 by MikePopoloski