MikePopoloski/slang #1211: Request: Add `SyntaxTree.isEquivalentTo(other_syntax_tree)` method

parker-research, 2025-01-14:
**Is your feature request related to a problem? Please describe.**
When using library to validate syntax tree modifications, it would be helpful to be able to use this method. This is very common in tests using slang/pyslang.

**Describe the solution you'd like**
Implement the method, as proposed.

**Describe alternatives you've considered**
I suppose you'd have to manually traverse the tree, currently.

MikePopoloski, 2025-01-15:
This isn't any different from just calling isEquivalentTo on the SyntaxTree::root() right? Not sure adding a wrapper method really adds much value here.

parker-research, 2025-01-15:
Oh okay, fair enough! Based on the bug discovered in https://github.com/MikePopoloski/pyslang/issues/406 (which was unclear to me as to whether it was a bug or undocumented behaviour), this makes sense.

closed 2025-01-15 by parker-research