MikePopoloski/slang #1221: [LRM-2012][lang-compliance] Operator overloading

likeamahoney, 2025-01-21:
Hi, all!

Operator overloading was introduced at SystemVerilog LRM 2012 (section 11.11) and was deprecated by LRM 2017 due to the fact that no one supported this feature in the tools and, accordingly, did not use it.

But it seems to me that this feature is quite convenient for building testbenches that widely use package structures, slang is powerful enough to support it, but not by default, but with some option, like `--depreaced-legacy`. Is support for this considered or is this not needed in `slang`?

MikePopoloski, 2025-01-22:
As far as I know no other tool ever implemented this. Sure, slang could add support, but who would ever write code using it if they couldn't simulate it or synthesize it or whatever?

likeamahoney, 2025-01-22:
> As far as I know no other tool ever implemented this. Sure, slang could add support, but who would ever write code using it if they couldn't simulate it or synthesize it or whatever?

`slang` (as possible fronted for other tools) could optionally inline bodies by substituting the function body for the overloaded operator, this would make the overloading implicit for tools (the tools would not even know about its presence), thus automatic support by other tools would be provided

MikePopoloski, 2025-01-22:
Hmm, I see. I suppose that makes sense, though I think such a project is a fair amount of work and I kind of doubt anyone would ever really use it in practice.

closed 2025-09-06 by MikePopoloski