MikePopoloski/slang #1222: Ternary sizing regression in presence of constants

povik, 2025-01-21:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
C++ slang

**Describe the bug**
Commit dd2addd6 introduced a regression where ternaries can be resolved to a constant of improper size.

**To Reproduce**
```
TEST_CASE("Test ternary operation sizing regression") {
    auto tree = SyntaxTree::fromText(R"(
module m;
  logic [3:0] w = (0) ? '0 : '0;
endmodule
)");

    Compilation compilation;
    compilation.addSyntaxTree(tree);
    NO_COMPILATION_ERRORS;

    auto& root = compilation.getRoot();
    auto& var = root.lookupName<VariableSymbol>("m.w");

    ast::EvalContext eval_ctx(ast::ASTContext(compilation.getRoot(),
                                              ast::LookupLocation::max));
    CHECK(var.getInitializer()->eval(eval_ctx).getBitstreamWidth() == 4);
}
```

**Additional context**
AST dump from another instance (not the same as the test above):
```
{
  "kind": "ConditionalOp",
  "type": "logic[7:0]",
  "conditions": [
    {
      "expr": {
        "kind": "BinaryOp",
        "type": "logic",
        "op": "Equality",
        "left": {
          "kind": "NamedValue",
          "type": "logic signed[31:0]",
          "symbol": "5621136432 width_p",
          "constant": "8"
        },
        "right": {
          "kind": "Conversion",
          "type": "logic signed[31:0]",
          "operand": {
            "kind": "IntegerLiteral",
            "type": "int",
            "value": "1",
            "constant": "1"
          },
          "constant": "1"
        },
        "constant": "1'b0"
      }
    }
  ],
  "left": {
    "kind": "UnbasedUnsizedIntegerLiteral",
    "type": "logic[7:0]",
    "value": "8'd0",
    "constant": "8'd0"
  },
  "right": {
    "kind": "UnbasedUnsizedIntegerLiteral",
    "type": "logic[7:0]",
    "value": "8'd0",
    "constant": "1'b0"
  },
  "constant": "1'b0"
}
```

MikePopoloski, 2025-01-24:
Fixed in e9107cf22f758c0d84c75362de185ca8a4cff77b

closed 2025-01-24 by MikePopoloski

povik, 2025-01-24:
Thanks!

Do you want to add a check for the constant width matching the type? I effectively have one in yosys-slang and that’s how I’ve been finding these cases.

MikePopoloski, 2025-01-24:
Yeah, I was thinking that myself. Just need to find the right place to do it.