MikePopoloski/slang #1230: python: Enable ruff linter for Python code

parker-research, 2025-02-03:
**Is your feature request related to the C++ slang project or the pyslang Python bindings?**
Pyslang Python bindings

**Is your feature request related to a problem? Please describe.**
The Python tests are written in good-but-not-quite-optimal modern Python. Turning on the ruff linter for the Python bindings would help remedy this. 

**Describe the solution you'd like**
ruff is a linter and an autoformatter. It is quite performant, and works well in CI workflows (i.e., as a GitHub Action).

**Describe alternatives you've considered**
Black, flake8, etc. are alternatives. Ruff seems to be the community's choice for new projects, and it's the one I prefer too! 

**Additional context**
Later on, enabling pyright or mypy in CI would be good too!

MikePopoloski, 2025-02-04:
flake8 and black are already enabled via pre-commit. I guess if you want to convert over to ruff I don't care that much but is it really that different?

parker-research, 2025-02-04:
Hmm, might be more an issue of just enabling the right Flake8 settings then. Examples of things I've seen that are disallowed by ruff in its mostly-default setup include:

* No `from pyslang import *` glob imports
* Return type annotations required (including for None-returning functions)
* Docstrings

MikePopoloski, 2025-04-01:
Closing this since there already is a linter and there's been no followup to the discussion.

closed 2025-04-01 by MikePopoloski

parker-research, 2025-04-07:
You asked "is it really that different?"

I mentioned the benefits, and was waiting for your approval/decline.

Is it okay if I make that switch?

MikePopoloski, 2025-04-07:
Yes, if you want to tweak the flake8 settings that's fine. If you really want ruff then it needs to work via pre-commit as well.