MikePopoloski/slang #1232: error vs warning in Python

renau, 2025-02-04:
I have this python code:
```
import pyslang

tree = pyslang.SyntaxTree.fromText("""
module trivial(input [2:0] a, input [2:0] b, output [1:0] c);
assign c = a ^ b;
endmodule;
""")

deng = pyslang.DiagnosticEngine

c = pyslang.Compilation()
c.addSyntaxTree(tree)
diags = c.getAllDiagnostics()

for err in diags:
    print("===\n")
    print(err.code)
    print("XX" if err.isError else "never printed")
    reports = deng.reportAll(c.sourceManager, [err])
    print(reports)
```

This Verilog has 1 WARNING (3 to 2 bits) and 1 ERROR (; after endmodule)

When I iterate over the diags, I would like to know which is an error and which is a warning. It always prints "XX". isError is always true.

How can I know it?

Also, if I try:
```
    print(deng.getSeverity(err.code, err.location))
```
I get this error:
```
Traceback (most recent call last):
  File "/Users/renau/projs/hagent/tmp/./slang2.py", line 19, in <module>
    print(deng.getSeverity(err.code, err.location))
          ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: getSeverity(): incompatible function arguments. The following argument types are supported:
    1. (self: pyslang.DiagnosticEngine, code: pyslang.DiagCode, location: pyslang.SourceLocation) -> pyslang.DiagnosticSeverity

Invoked with: DiagCode(WidthTruncate), SourceLocation(1, 74)
```

How should I call the getSeverity?

MikePopoloski, 2025-02-05:
The `; after endmodule` diagnostic is a warning, not an error, which is why `isError` does not return true.

`getSeverity` is not working for you because `deng` is the static type of `DiagnosticEngine`, not an instance. You need to call the constructor to get a real instance, for example:
```
deng = pyslang.DiagnosticEngine(tree.sourceManager)
```

MikePopoloski, 2025-02-05:
`isError` is also a function, so you need parentheses when calling it:
```
if err.isError():
    print("XX")
```
I guess Python is evaluating the function object itself as always being true because you aren't including the parentheses.

renau, 2025-02-05:
Thanks, closing the issue

closed 2025-02-05 by renau