MikePopoloski/slang #1238: Python docs/examples: Add example of using visitor callback function

parker-research, 2025-02-11:
**Is your feature request related to the C++ slang project or the pyslang Python bindings?**
Pyslang Python bindings

**Is your feature request related to a problem? Please describe.**
The [pyslang SyntaxNode docs](https://sv-lang.com/pyslang/pyslang.SyntaxNode.html) present the following method:

[image]

I can't figure out how to use it as a visitor/transformer. An example in `/pyslang/examples/` and/or the docs would be very helpful! 

**Additional context**
If you can give a half-baked example in this issue, I can help prepare a PR.

MikePopoloski, 2025-02-12:
The test_visitors.py file already makes use of this method. Can you expand on those if you want them to do something else?

MikePopoloski, 2025-04-01:
Closing since you've opened #1291 which covers this.

closed 2025-04-01 by MikePopoloski