MikePopoloski/slang #1239: Python docs/examples: Add example of visiting and transforming AST

parker-research, 2025-02-11:
**Is your feature request related to the C++ slang project or the pyslang Python bindings?**
Pyslang Python bindings

**Is your feature request related to a problem? Please describe.**
I think the following would make a good example:

Create an example Python script which visits all SyntaxNodes, and renames all signals it encounters to be suffixed with "_visited". 

Such an example would demonstrate how to use pyslang to transform a syntax tree.

**Describe the solution you'd like**
A clear example in Python would be helpful. 

It could even be written in the style of a unit test, with the following input and output:

### Example Input

```verilog
module RefModule (
  input in1,
  input in2,
  output logic out
);

  assign out = in1 & ~in2;

endmodule
```

### Example Output

```verilog
module RefModule (
  input in1_visited,
  input in2_visited,
  output logic out_visited
);

  assign out_visited = in1_visited & ~in2_visited;

endmodule
```

**Additional context**
I'd be happy to help finish this in a PR if you can just give a crude example of how this is accomplished.

MikePopoloski, 2025-02-12:
I don't think this is currently possible to do with the Python API. APIs that mutate or create nodes aren't exposed in the bindings, so you'd have to start by adding those. There are some examples of using the C++ API to do stuff like this in VisitorTests.cpp

MikePopoloski, 2025-04-01:
Closing since there is now #1271

closed 2025-04-01 by MikePopoloski