MikePopoloski/slang #1242: Issues when processing with json format

jiao-x, 2025-02-13:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
C++ slang 

**Describe the bug**
When I generate a json file from RTL to process, I recursively process the different 'kind' types. Why is there no annotation for kind, which makes it more difficult to process.

"conditions": [
                    {
                      "expr": {
                        "kind": "UnaryOp",
                        "type": "logic",
                        "op": "BitwiseNot",
                        "operand": {
                          "kind": "NamedValue",
                          "type": "logic",
                          "symbol": "6283438802072 rst_n"
                        }
                      }
                    }
                  ],

MikePopoloski, 2025-02-13:
I do not know what you're trying to say here. I see "kind" elements in the example you posted.

jiao-x, 2025-02-14:
The pattern I've observed in json is that every dictionary has a "kind" element, but the one I've marked below doesn't.

"conditions": [
{
“kind”:❓❓❓
"expr": {
"kind": "UnaryOp",
"type": "logic",
"op": "BitwiseNot",
"operand": {
"kind": "NamedValue",
"type": "logic",
"symbol": "6283438802072 rst_n"
}
}
}
],

MikePopoloski, 2025-02-14:
Elements only have a "kind" attribute when it's necessary to distinguish between different types of polymorphic elements. For example, there are many different kinds of expressions, so each expression instance carries a "kind" attribute with it. The array you've posted here is always an array of Conditions so the kind would always be the same, so there's no reason to serialize it.

MikePopoloski, 2025-02-17:
Closing since there's been no follow up.

closed 2025-02-17 by MikePopoloski