MikePopoloski/slang #1244: Some possible improvements for -Wint-bool-conv

jcurtiss8086, 2025-02-17:
I've noticed that -Wint-bool-conv is not flagged in assert and cover properties when a multi-bit variable is specified stand-alone without any logical operators, but this case is flagged for if-statements.

Also, in the case of an equation like 'a & b' where the result is still multi-bit , both the if-statement and the assert/cover properties do not flag the warning.

Are these cases that could be covered by this warning as well?

```
coerce_int_bool.sv:6:13: warning: implicit conversion from 'logic[1:0]' to boolean value [-Wint-bool-conv]
        if (a) begin
            ^
coerce_int_bool.sv:15:36: warning: implicit conversion from 'logic[1:0]' to boolean value [-Wint-bool-conv]
    cover property (@(posedge clk) a && b);
                                   ^
coerce_int_bool.sv:15:41: warning: implicit conversion from 'logic[1:0]' to boolean value [-Wint-bool-conv]
    cover property (@(posedge clk) a && b);
                                        ^

```
```Systemverilog
module top;
    logic clk;
    logic [1:0] a, b;

    always_comb begin
        if (a) begin
       end 
       if (a & b) begin
       end
    end

    assert property (@(posedge clk) a |-> b);
    cover property (@(posedge clk) a);
    cover property (@(posedge clk) a & b);
    cover property (@(posedge clk) a && b);
endmodule
```

MikePopoloski, 2025-02-17:
The warning deliberately does not fire for bitwise AND, XOR, or right shift, since it's very common to test for flags or high bits set like this:
```
logic [3:0] a;
if (a & 'b0100) begin end
if (a >> 2) begin end
```
Do you see this as a potential source of bugs that you're hoping to catch?

It makes sense to warn for boolean expressions in sequence / properties though, I can add that.

jcurtiss8086, 2025-02-17:
I don't have any explicit examples -- your explanation sounds like a good reason to not include the bitwise operations.

MikePopoloski, 2025-03-09:
Added in 83a09abe0a7810ca47e2d100c5f417c3d81bb968

closed 2025-03-09 by MikePopoloski