MikePopoloski/slang #1248: [bug][feature-request][analysis] Eval-based analysis restrictions

likeamahoney, 2025-02-18:
Hi, I am inspired and excited about the advent of static analysis in `slang`.

It is a great feature!

But condition expression eval-based approach turns out to be quite limited.

I suppose analysis engine needs a more complex condition evaluator mechanism like for example `clang CSA` have (which is based on variable range constraints by `clang ConstraitManager`). 

Because without it `slang` cannot perform correct analysis of such simple cases:

```verilog
module m();

reg w;

always_comb begin
    if (w == w) // Literally no latch - always equals to `true` but slang analysis engine reports this
        w = 1'b0;
end

endmodule 
```

Is there a plan to introduce something like this or how is it planned to develop the analysis engine?

likeamahoney, 2025-02-18:
or such example where `slang` produce inferred latch false positives:

```verilog
module m();

reg w, r;

always_comb begin
    r = 1'b1;
    if (r == 1'b1) // Literally no latch - always equals to `true` but slang analysis engine reports this
        w = 1'b0;
end

endmodule 
```

MikePopoloski, 2025-02-19:
The feature is definitely not complete yet so I wouldn't read too much into it, but sure, the constant evaluation can be improved to reduce false positives. I'm not sure how far to take it; there's no definition in the LRM for what constitutes a latch and at what level of constant evaluation should be done to find out, so the main goal is to help people find bugs in their real world code and not try to satisfy weird edge cases that can be dreamt up for testing purposes.

MikePopoloski, 2025-04-01:
Closing this since it's not directly actionable. I agree with the general sentiment: there are plenty of improvements to be made to the analysis flow, and specific issues can be opened to address them.

closed 2025-04-01 by MikePopoloski