MikePopoloski/slang #1255: Linking hazard because layout of BumpAllocator sensitive to NDEBUG macro

povik, 2025-02-25:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
C++ slang

**Describe the bug**

Commit [dd222bd](https://github.com/MikePopoloski/slang/commit/dd222bd98256621dfba9ae9a091103502d5d9f1d) introduced the `bool freeze;` field on BumpAllocator which is conditionally included depending on the `SLANG_ASSERT_ENABLED` macro, which in turn can depend on `NDEBUG`.

This causes issues if slang is compiled in release mode, but then is linked into a non-cmake application which does not define `NDEBUG`.

**To Reproduce**

**Additional context**
Not sure what the best practice would be here -- if external linkage to slang requires particular defines, perhaps they should be placed into the installed sv-lang.pc, or the layout of structs/classes should not depend on defines at all.

MikePopoloski, 2025-02-25:
This has always been a problem; for example, SourceLocation's layout depends on the `DEBUG` define. I think it's probably a lot more likely to run into issues with `NDEBUG` though since that's a standard define and it's probably fairly common to build slang as a release shared lib and your application consuming it in debug mode.

It is probably true that the pkgconfig file is deficient; it was a contribution from someone in the past and I don't have anything testing that it works. Contributions to fix it are welcome. A pragmatic improvement might also be to use `DEBUG` (probably should rename to `SLANG_DEBUG`) instead of `NDEBUG` since installing slang in release mode is probably the most common usage.

MikePopoloski, 2025-02-25:
Yes, it should. Well, it does for `DEBUG` as it's listed as a `PUBLIC` definition that comes along with the target when you link to it via CMake. Perhaps it does not for NDEBUG? Maybe NDEBUG is just too problematic to use in general since it's implicit behavior from the compiler and not known to the build system.

MikePopoloski, 2025-02-25:
I suppose this also comes up with the Boost single header vs system include. If you build slang with the vendored single header but don't carry along the preprocessor define then your application can pick up a system installed boost instead. You might get lucky and have the layouts be ABI compatible or you may crash in bizarre ways due to ODR violations.

povik, 2025-02-25:
Yeah, I've been working around this by forcing the vendored header: https://github.com/povik/yosys-slang/blob/master/Makefile#L30

MikePopoloski, 2025-03-13:
Fixed the NDEBUG usage in e925ae1e8d72335d8a5ac3b9da9a6067e6afb913

Hopefully that's sufficient.

closed 2025-03-17 by MikePopoloski

povik, 2025-03-18:
Thanks Mike! @mattyoung1 has reported in the linked thread the define change has helped. I've been also looking at ways to extend the sv-lang.pc generation but I was disappointed with the cmake support for this use case. This comment on the cmake tracker paints a grim picture: https://gitlab.kitware.com/cmake/cmake/-/issues/22621#note_1106762