MikePopoloski/slang #1257: Flag for maximum compatibility

povik, 2025-02-25:
**Is your feature request related to the C++ slang project or the pyslang Python bindings?**
C++ slang

**Is your feature request related to a problem? Please describe.**
I see users of yosys-slang piling compatibility flags on top of each other (typically `--compat=vcs` and other flags they find) to parse existing codebases which they don't want to lint or modify in any way. It's not that they want compatibility with a specific tool, they want to make the reader as relaxed as they can make it. I think it's best if there's a one-size-fits-all way to do this.

**Describe the solution you'd like**
I suggest `--compat=max`, `--relax-all`, or another flag which turns all downgradable errors into warnings, and enables all available tweaks within slang for accepting non-compliant input.

I'm happy to contribute this once you agree to it.

MikePopoloski, 2025-02-25:
Yes, I'm fine with something like `--compat=max` or maybe `--compat=all`. There are a handful of downgradable errors that are particularly sketchy that you might want to give extra thought to, namely the ones that allow unknown modules, duplicate module definitions, and unknown system calls, but as long as we're putting this behind a flag that makes it clear to the user what they're getting themselves into I don't care that much.

MikePopoloski, 2025-02-25:
Also I'm happy to bump any of the max depth / max size compilation limits if someone is hitting them on a real design. The defaults were set somewhat arbitrarily.

povik, 2025-02-25:
OK, thanks

> There are a handful of downgradable errors that are particularly sketchy that you might want to give extra thought to, namely the ones that allow unknown modules, duplicate module definitions, and unknown system calls

That's a good point. I'm not sure about unknown system calls (for a synthesis tool ignoring those sounds innocuous), but unknown or duplicate modules might be going too far with accepting anything.