MikePopoloski/slang #1258: Segfaults with assert properties

jcurtiss8086, 2025-02-26:
Starting somewhere between 780e4783f070e4f2ff07e0387d986bef2d3014bb and f41e439995182e0ef4ff8d9e9a6f101cf9d7c4a0 inclusive, the following example segfaults.  Builds are using gcc 12.2.0.  Debug builds do not trigger an ICE.

```
module top;
    logic clk, rst, a;

    lbl: assert property(@(posedge clk) disable iff (rst) a);
endmodule
```

The unittests binary passes, but if I copy the following example out of MemberTests.cpp to run standalone, it also sefaults.
```
module M #(
    A = 1
);
logic clk, rst;

property myprop(k);
   @(posedge clk) disable iff(rst !== 0) k > 0;
endproperty

genvar k;
for (k=1; k < 4; k++) begin: m
    if (A)
        label1: assert property(myprop(k));
    else
        label2: assert property(myprop(k));

    if (A)
        label3: assert property(myprop(k))
                else $error("assert failed");
    else
        label4: assert property(myprop(k));

    if (A) begin
        label5: assert property(myprop(k));
    end else
        label6: assert property(myprop(k));
end

endmodule
```

MikePopoloski, 2025-02-27:
Fixed in #1259

closed 2025-02-27 by MikePopoloski

jcurtiss8086, 2025-02-27:
Both of the above examples are still producing segfaults for my local build.  The cmake version and command are:
```
/path/to/cmake/3.29.1/bin/cmake -DCMAKE_CXX_COMPILER=/path/to/gcc/12.2.0/bin/g++ -DCMAKE_BUILD_TYPE=Release 
make -j16
```

likeamahoney, 2025-02-27:
> Both of the above examples are still producing segfaults for my local build. The cmake version and command are:
> 
> ```
> /path/to/cmake/3.29.1/bin/cmake -DCMAKE_CXX_COMPILER=/path/to/gcc/12.2.0/bin/g++ -DCMAKE_BUILD_TYPE=Release ..
> make -j16
> ```

Bug fix https://github.com/MikePopoloski/slang/pull/1259 was removed by https://github.com/MikePopoloski/slang/commit/d9672777a57f18f6b2611fb1c49f8f3517005ab7#diff-5a70d0d0d6b298c84fb8b078f97ebd51b4df487cce4a47036caba8d5eab76546R301

MikePopoloski, 2025-02-27:
Yeah sorry, was a bad merge on my part.