MikePopoloski/slang #1265: Documentation

nitinhariharan, 2025-02-28:
Pyslang is great! Could you please provide any documentation? It is quite difficult to use.

nitinhariharan, 2025-02-28:
The website documentation has no proper example and there are lot of functions in the module.
please help!
Thanks

MikePopoloski, 2025-03-01:
There are examples in the repo under the pyslang directory. @parker-research is working on some docs for the website.

nitinhariharan, 2025-03-01:
Great 
looking forward for it 
Thanks !

parker-research, 2025-03-03:
Yeah, at least a bit in progress in #1253.

parker-research, 2025-03-17:
I think this could maybe be marked complete now? Docs site is live: https://sv-lang.com/pyslang/

Obviously far more can be done there, but I think that specific requests should go into separate issues.

closed 2025-03-17 by MikePopoloski