MikePopoloski/slang #1271: Pyslang: Add bindings for `SyntaxRewriter`

parker-research, 2025-03-03:
**Is your feature request related to the C++ slang project or the pyslang Python bindings?**
Pyslang Python bindings

**Is your feature request related to a problem? Please describe.**
It seems that the Python bindings for the `slang::syntax::SyntaxRewriter` don't exist yet. Porting these to Python would be a huge benefit.

**Describe the solution you'd like**
Port the `slang::syntax::SyntaxRewriter` bindings.

**Describe alternatives you've considered**
Learn C++ better ;)

**Additional context**
Seems that the visitor is already ported (as in `test_visitors.py`), but that the visitor system is exclusively read-only currently.

parker-research, 2025-03-17:
I've been toying with this a bit, albeit extremely unsuccessfully. I'm super open to any advice/opinions/guidance.

Can you help me work out what we think a reasonable API would be in Python? This is sorta what I was thinking so far:

```
Call SyntaxTree(...).rewrite(handler_function_f)

Then, `rewrite.f` is described by:

        "Rewrites a syntax tree by applying transformations specified in callback `f`.\n\n"
        "The callback function `f` takes two arguments: the current syntax node "
        "being visited, and a `rewriter` argument of type `SyntaxRewriter`. "
        "The callback can use the following methods of `rewriter` to modify the tree:\n"
        "- rewriter.remove(old_node): Removes the given node from the tree\n"
        "- rewriter.replace(old_node, new_node): Replaces old_node with new_node\n"
        "- rewriter.insert_before(old_node, new_node): Inserts new_node before old_node\n"
        "- rewriter.insert_after(old_node, new_node): Inserts new_node after old_node\n"
        "\n"
        "Returns a new transformed SyntaxTree object.";
```

### Example Tests

```python
import pytest
import pyslang
from pyslang import SyntaxTree, SyntaxKind

def test_rewriter_handler_function_called_with_right_args():
    tree = SyntaxTree.fromText("""
        module m;
            int i;
            logic l;
        endmodule
    """, "test.sv")

    handler_tracker = {"call_count": 0}

    def handler(*args, **kwargs):
        assert len(args) == 2
        assert isinstance(args[0], pyslang.SyntaxNode)
        assert isinstance(args[1], pyslang.SyntaxRewriter)
        assert len(kwargs) == 0

        handler_tracker["call_count"] += 1

    assert handler_tracker["call_count"] == 0
    result = pyslang.rewrite(tree, handler)
    assert result is not None
    assert isinstance(result, SyntaxTree)

    assert handler_tracker["call_count"] > 0, "Handler should have been called at least once"
    assert handler_tracker["call_count"] >= 4, "Handler should have been called at least 4 times"

def test_rewriter_with_no_changes():
    tree = SyntaxTree.fromText("""
        module m;
            int i;
            logic l;
        endmodule
    """, "test.sv")
    expected = SyntaxTree.fromText("""
        module m;
            int i;
            logic l;
        endmodule
    """, "test.sv")
    result = pyslang.rewrite(tree, lambda _node, _rewriter: None)
    assert result is not None
    assert result.root.isEquivalentTo(expected.root)

def test_rewriter_remove():
    print("STARTING TEST")
    tree = SyntaxTree.fromText("""
        module m;
            int i;
            logic l;
        endmodule
    """, "test.sv")
    expected = SyntaxTree.fromText("""
        module m;
            logic l;
        endmodule
    """, "test.sv")

    check_func_called = {'called': False}
    
    def remove_int_var(node: pyslang.SyntaxNode, rewriter: pyslang.SyntaxRewriter):
        assert isinstance(node, pyslang.SyntaxNode)
        assert isinstance(rewriter, pyslang.SyntaxRewriter)

        check_func_called['called'] = True

        if node.kind == SyntaxKind.DataDeclaration:
            if node[0].kind == SyntaxKind.DataType and node[0][0].rawText == "int":
                node.remove(node)

    assert check_func_called['called'] is False

    result = pyslang.rewrite(tree, remove_int_var)
    assert check_func_called['called'] is True
    assert result is not None
    assert result.root.isEquivalentTo(expected.root)
```

MikePopoloski, 2025-03-20:
Yes, that API seems reasonable to me.

PR #1307 (2025-04-12): pyslang: Create bindings for SyntaxRewriter
Massive project - implementing bindings and tests for SyntaxRewriter. Finally done. 

There seems to be one quirk - the check at the end of the `test_rewriter_nested()` test (see the `# FIXME` comment). Would really appreciate it if you could help me take a look at that and see why `assert result.root.isEquivalentTo(expected.root) is True` fails, but checking string equality (neglecting whitespaces) passes. 

Fixes #1271
closed 2025-04-12 by MikePopoloski