MikePopoloski/slang #1274: referenced AST node not reachable by AST walker?

tdp2110, 2025-03-12:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
C++ slang 

**Describe the bug**
An example with a seemingly missing instance reachable from the AST walkers

**To Reproduce**

Consider 
```
interface vif #(PARAM = 540)
(
  input clk,
  input rstn
);

endinterface

class cls #(int PARAM = 540,
            type T = virtual vif#(PARAM)
            );
   T m_vif;
   task wait_for_rst();
     wait(m_vif.rstn == 0);
   endtask

endclass

module top;
   cls c = new;
endmodule
```

This has the following (error-free) ast-json:

```
{
  "design": {
    "name": "$root",
    "kind": "Root",
    "addr": 3109020978240,
    "members": [
      {
        "name": "",
        "kind": "CompilationUnit",
        "addr": 3109021721824,
        "members": [
          {
            "name": "cls",
            "kind": "GenericClassDef",
            "addr": 3109021693424,
            "specializations": [
              {
                "name": "cls",
                "kind": "ClassType",
                "addr": 3109021723552,
                "members": [
                  {
                    "name": "PARAM",
                    "kind": "Parameter",
                    "addr": 3109021723800,
                    "type": "int",
                    "initializer": {
                      "kind": "IntegerLiteral",
                      "type": "int",
                      "value": "540",
                      "constant": "540"
                    },
                    "value": "540",
                    "isLocal": false,
                    "isPort": true,
                    "isBody": false
                  },
                  {
                    "name": "T",
                    "kind": "TypeParameter",
                    "addr": 3109021724248,
                    "type": "vif#(PARAM=540)",
                    "isLocal": false,
                    "isPort": true,
                    "isBody": false
                  },
                  {
                    "name": "m_vif",
                    "kind": "ClassProperty",
                    "addr": 3109021725800,
                    "type": "cls::T",
                    "lifetime": "Automatic",
                    "visibility": "Public"
                  },
                  {
                    "name": "wait_for_rst",
                    "kind": "Subroutine",
                    "addr": 3109021726152,
                    "members": [
                      {
                        "name": "this",
                        "kind": "Variable",
                        "addr": 3109021726440,
                        "type": "cls",
                        "lifetime": "Automatic",
                        "flags": "const,compiler_generated"
                      }
                    ],
                    "returnType": "void",
                    "defaultLifetime": "Automatic",
                    "subroutineKind": "Task",
                    "body": {
                      "kind": "Wait",
                      "cond": {
                        "kind": "BinaryOp",
                        "type": "logic",
                        "op": "Equality",
                        "left": {
                          "kind": "Conversion",
                          "type": "logic[31:0]",
                          "operand": {
                            "kind": "NamedValue",
                            "type": "logic",
                            "symbol": "3109021733376 rstn"
                          }
                        },
                        "right": {
                          "kind": "Conversion",
                          "type": "logic[31:0]",
                          "operand": {
                            "kind": "IntegerLiteral",
                            "type": "int",
                            "value": "0",
                            "constant": "0"
                          },
                          "constant": "32'd0"
                        }
                      },
                      "stmt": {
                        "kind": "Empty"
                      }
                    },
                    "visibility": "Public",
                    "arguments": [
                    ]
                  }
                ],
                "isAbstract": false,
                "isInterface": false,
                "isFinal": false,
                "genericClass": "3109021693424 cls",
                "implements": [
                ]
              }
            ]
          }
        ]
      },
      {
        "name": "top",
        "kind": "Instance",
        "addr": 3109021722648,
        "body": {
          "name": "top",
          "kind": "InstanceBody",
          "addr": 3109021722120,
          "members": [
            {
              "name": "c",
              "kind": "Variable",
              "addr": 3109021722304,
              "type": "cls",
              "initializer": {
                "kind": "NewClass",
                "type": "cls",
                "isSuperClass": false
              },
              "lifetime": "Static"
            }
          ],
          "definition": "3109020593664 top"
        },
        "connections": [
        ]
      }
    ]
  },
  "definitions": [
    {
      "name": "top",
      "kind": "Definition",
      "addr": 3109020593664,
      "defaultNetType": "3109020976000 wire",
      "definitionKind": "Module",
      "defaultLifetime": "Static",
      "unconnectedDrive": "None"
    },
    {
      "name": "vif",
      "kind": "Definition",
      "addr": 3109020583680,
      "defaultNetType": "3109020976000 wire",
      "definitionKind": "Interface",
      "defaultLifetime": "Static",
      "unconnectedDrive": "None"
    }
  ]
}
```

Class `cls` is specialized with the default specialization. It has a member of type `T= vif#(.PARAM(540))`. 
No instance for this interface is referenced in the Json dump. Further, it does not appear reachable by an ASTVisitor (done in C++, not shown). However, by examining this `m_vif` member I can find an instance of `vif` by walking its type. This instance has parent scope the `ClassType` `cls`, but it does not appear as a member of that class (lol wut?). This instance has json dump:

```
"body": {
  "name": "vif",
  "kind": "InstanceBody",
  "source_file": "vif_ref.sv",
  "source_line": 1,
  "source_column": 11,
  "addr": 3655254323344,
  "members": [
    {
      "name": "PARAM",
      "kind": "Parameter",
      "source_file": "vif_ref.sv",
      "source_line": 1,
      "source_column": 17,
      "addr": 3655254323528,
      "type": "logic signed[31:0]",
      "initializer": {
        "kind": "Conversion",
        "type": "logic signed[31:0]",
        "operand": {
          "source_file_start": "vif_ref.sv",
          "source_file_end": "vif_ref.sv",
          "source_line_start": 11,
          "source_line_end": 11,
          "source_column_start": 35,
          "source_column_end": 40,
          "kind": "NamedValue",
          "type": "int",
          "symbol": "3655254322328 PARAM",
          "constant": "540"
        },
        "constant": "540"
      },
      "value": "540",
      "isLocal": false,
      "isPort": true,
      "isBody": false
    },
    {
      "name": "clk",
      "kind": "Port",
      "source_file": "vif_ref.sv",
      "source_line": 3,
      "source_column": 9,
      "addr": 3655254331048,
      "type": "logic",
      "direction": "In",
      "internalSymbol": "3655254331408 clk"
    },
    {
      "name": "clk",
      "kind": "Net",
      "source_file": "vif_ref.sv",
      "source_line": 3,
      "source_column": 9,
      "addr": 3655254331408,
      "type": "logic",
      "netType": {
        "name": "wire",
        "kind": "NetType",
        "source_file": "",
        "source_line": 0,
        "source_column": 0,
        "addr": 3655252201120,
        "type": "logic"
      }
    },
    {
      "name": "rstn",
      "kind": "Port",
      "source_file": "vif_ref.sv",
      "source_line": 4,
      "source_column": 9,
      "addr": 3655254331776,
      "type": "logic",
      "direction": "In",
      "internalSymbol": "3655254331904 rstn"
    },
    {
      "name": "rstn",
      "kind": "Net",
      "source_file": "vif_ref.sv",
      "source_line": 4,
      "source_column": 9,
      "addr": 3655254331904,
      "type": "logic",
      "netType": {
        "name": "wire",
        "kind": "NetType",
        "source_file": "",
        "source_line": 0,
        "source_column": 0,
        "addr": 3655252201120,
        "type": "logic"
      }
    }
  ],
  "definition": "3655253452800 vif"
},
"connections": [
]
```

My question is, why is this Instance not reachable by the AST walker, even though it is referenced by the AST?

**Additional context**
Thanks again, you rock! 🥇

MikePopoloski, 2025-03-13:
A virtual interface type creates a dummy instance, both to verify that the declaration is valid given the provided parameter values and so that there are symbols to access when you dot into the variable. It's not a real instance so it doesn't get added to any members list, but we do set the parent pointer since various places in the code assume the invariant that everything has a parent pointer.

I assume you're accessing this via the `iface` member of the `VirtualInterfaceType`. The default ASTVisitor doesn't reach into that member but I suppose it could; it depends on what you're trying to accomplish.

tdp2110, 2025-03-13:
Thanks for explaining, Mike.