MikePopoloski/slang #1283: [slang][analysis] Multiclocked property doesn't warned

likeamahoney, 2025-03-19:
All examples from LRM 16.16 section clock resolution are warned by slang analysis correctly except one:

```
module top (input logic a, b, c, clk);
    property q1;
        $rose(a) |-> ##[1:5] b;
    endproperty

    property q2;
        @(posedge clk) q1;
    endproperty

    always @(negedge clk)
    begin
    a1: assert property ($fell(c) |=> q1);
        // illegal: multiclocked property with contextually
        // inferred leading clocking event
    end

endmodule
```

This example of property are compliant due to `slang` but LRM recommends to warn on that according to multiclock (`always - negedge clk` mixes with property `q2` leading `posedge clk`). Why is there no error here - because of the implication? Is this an error in the LRM?

MikePopoloski, 2025-03-20:
Did you copy the example incorrectly? `q2` is not used in your example.

Anyway, I believe this is an error in the LRM. There is no rule in the text that would make this illegal. Properties are allowed to be multiclocked. No commercial tool errors on this either.

closed 2025-03-20 by MikePopoloski

likeamahoney, 2025-03-20:
> Did you copy the example incorrectly? `q2` is not used in your example.
> 
> Anyway, I believe this is an error in the LRM. There is no rule in the text that would make this illegal. Properties are allowed to be multiclocked. No commercial tool errors on this either.

Thank you for response!