MikePopoloski/slang #1285: Spurious mutliple assignments error with interfaces and instance caching

povik, 2025-03-22:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
C++ slang

**Describe the bug**
Instance caching causes spurious "multiple assignments" diagnostic.

**To Reproduce**
Take the following source
```
interface bus();
	logic w;
endinterface

module m1(bus intf);
	assign intf.w = 0;
endmodule

module top();
	bus intf1();
	bus intf2();

	m1 a(.intf(intf1));
	m1 b(.intf(intf2));
endmodule
```
it triggers
```
Top level design units:
    top

  in instance: top.intf2
test_canonical.sv:3:34: error: cannot have multiple continuous assignments to variable 'b'
        modport primary(input a, output b, input clk);
                                        ^
note: from 'top.b' and 'top.b'
```
which goes away with `--disable-instance-caching`

povik, 2025-03-23:
With the standard slang main I was only able to reproduce it after I inserted

```
auto visitor = slang::ast::makeVisitor();
for (auto instance : compilation->getRoot().topInstances)
    instance->visit(visitor);
```

just below the creation of compilation

```
compilation = driver.createCompilation();
```

likeamahoney, 2025-03-23:
@povik  Hi! I encountered this in visitor in my code, when visiting interfaces, modports, packages, and some hier refs etc.

This is related to this [caching code](https://github.com/MikePopoloski/slang/blob/389fab2ed8731f3beb3aa5c4cdb86cd2a27b07ee/source/ast/ElabVisitors.h#L606) in slang. Perhaps All this is due to the fact that in your visitors there is a repeated elaboration and because of the [cache](https://github.com/MikePopoloski/slang/blob/389fab2ed8731f3beb3aa5c4cdb86cd2a27b07ee/source/ast/ElabVisitors.h#L640), it considers the same values drives as different and a conflict occurs. As an option, it is to use a template parameter from [PR](https://github.com/MikePopoloski/slang/pull/1262) in your visitors to visit without elaboration.

povik, 2025-03-23:
I understand it's related to instance caching (after all it goes away with `--disable-instance-caching`), but what surprises me is that it's triggered by applying an empty visitor with no user code. Either it's a slang bug, or it's misuse of the slang API on my part. If it's misuse I'd like to get a clear statement of why so that I can use slang confidently moving forward.

MikePopoloski, 2025-03-24:
Fixed in 41c39ecd5ef9763a49fa4267bfd57278a373c892