MikePopoloski/slang #1288: Relative includes are not relative to current working directory

AndreasLoow, 2025-03-30:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**

C++ slang

**Describe the bug**

When describing what `include does, the SystemVerilog-2023 standard states the following:

> When the filename is enclosed in double quotes ("filename"), for a relative path the compiler’s
current working directory, and optionally user-specified locations are searched.

It seems that slang does not accurately implement this requirement.

**To Reproduce**

```text
> cat examples/test.sv 
`include "examples/empty.sv"
> cat examples/empty.sv 
> slang examples/test.sv 
examples/test.sv:1:10: error: 'examples/empty.sv': No such file or directory
`include "examples/empty.sv"
         ^~~~~~~~~~~~~~~~~~~
warning: no top-level modules found in design [-Wmissing-top]

Build failed: 1 error, 1 warning
```

**Additional context**

None.

AndreasLoow, 2025-03-30:
Some more context: thought about it a little more, inclusion relative to the current file seems more useful than the current working directory. But I guess the standard specifies the less useful behaviour so what to do. Seems that Icarus solves this by a flag:

```text
       -grelative-include|-gno-relative-include
               Enable or disable (default) adding the local files directory to the beginning of
               the include file search path. This allows files to be included relative to the
               current file not the more common files are only found in the working directory or
               in the specified include file search path.
```

MikePopoloski, 2025-03-30:
Yeah, assuming that all includes are relative to whatever the current working directory of the compiler happens to be by default seems completely bananas which is why I didn't do that. You can easily get that behavior if you want it though by just adding `-I./` to the command line.

AndreasLoow, 2025-03-31:
> You can easily get that behavior if you want it though by just adding `-I./` to the command line.

But that just adds more paths to the search path, it does not remove the ones already there, so that does not give us the behaviour suggested by the standard. See below.

```text
> cat examples/empty.sv 
> cat examples/test1.sv 
`include "examples/empty.sv"
> cat examples/test2.sv 
`include "empty.sv"
```

Adding `-I.` does, indeed, make the old test pass:

```
> slang examples/test1.sv 
examples/test1.sv:1:10: error: 'examples/empty.sv': No such file or directory
`include "examples/empty.sv"
         ^~~~~~~~~~~~~~~~~~~
warning: no top-level modules found in design [-Wmissing-top]

Build failed: 1 error, 1 warning
> slang -I. examples/test1.sv
warning: no top-level modules found in design [-Wmissing-top]

Build succeeded: 0 errors, 1 warning
```

But, at the same time, it makes too many things pass. E.g., according to the behaviour suggested by the standard, this should fail:

```
> slang -I. examples/test2.sv
warning: no top-level modules found in design [-Wmissing-top]

Build succeeded: 0 errors, 1 warning
```

MikePopoloski, 2025-03-31:
I'm struggling to understand what you're asking for here. Do you want a flag that makes more designs fail to parse by default? What would be the utility of such a flag? Who would use it?

AndreasLoow, 2025-04-01:
Sorry for the confusion, I was simply pointing out that adding `-I.` does not give the same behaviour as the behaviour specified by the standard (which is what I read your previous comment to say, but on a second read it might not have been what you meant). I opened this issue because I thought one of the main goals of slang was to implement the standard as accurately as possible (which is the reason I went with slang for my current use-case rather than some other tool), but if it's an intentional design decision to diverge from the standard here I guess that resolves this issue :). Anyhow, thanks for clarifying.

MikePopoloski, 2025-04-01:
That is one of the goals of slang, yes, but there are others, such as trying to maintain compatibility with commercial tools, providing a usable experience out of the box, etc. There are times where the LRM is either vague, misleading, or downright wrong about certain things, in which case slang needs to pick a better behavior based on those other priorities.

As far as LRM compatibility goes, you can read this thread on the subject where the committee discusses relative include paths and indicates that the intention was to follow C compiler behavior (which is what slang does) and that the existing wording is probably a mistake to be fixed: https://accellera.mantishub.io/view.php?id=2339