MikePopoloski/slang #1292: Internal compiler error in IntervalMap with looped array assignments in always_ff

hankhsu1996, 2025-03-31:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**  
C++ slang

**Describe the bug**  
Running slang on a simple module crashes with an internal compiler error in `IntervalMap.h`. The issue seems to relate to array assignments within a `for` loop inside an `always_ff` block.

**To Reproduce**  
Input file (`for_in_ff.sv`):
```systemverilog
module Test (
  input  logic clk,
  input  logic in,
  output logic out
);

  logic internal[4];

  assign internal[0] = in;
  assign out = internal[3];

  always_ff @(posedge clk) begin
    internal[1] <= internal[0];
    for (int stage = 2; stage <= 3; stage++) begin
      internal[stage] <= internal[stage-1];
    end
  end

endmodule
```

Command:
```bash
slang for_in_ff.sv
```

Output:
```
Top level design units:
    Test

internal compiler error: Assertion 'valid()' failed
  in file /home/hankhsu/workspace/c++/slang/source/../include/slang/util/IntervalMap.h, line 935
  function: slang::IntervalMap<TKey, TValue, N>::overlap_iterator& slang::IntervalMap<TKey, TValue, N>::overlap_iterator::operator++() [with TKey = long unsigned int; TValue = const slang::ast::Expression*; unsigned int N = 5]
```

**Additional context**  
- Slang version: 8.0.142+98534e59
- Platform: WSL Ubuntu 22.04
- Compiler: Clang 19
- Reproducible: Always

MikePopoloski, 2025-04-02:
Fixed in a683a49a18cee3fffea7bd5ad6a5345764411ddf

closed 2025-04-02 by MikePopoloski

hankhsu1996, 2025-04-02:
That is fast. thanks!!