MikePopoloski/slang #1293: -Wconversion flag misses width checks for literals and constant expressions

hankhsu1996, 2025-03-31:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**  
C++ slang

**Describe the bug**  
With `-Wconversion` enabled, slang correctly warns about width truncation or expansion for variables and concatenations of variables, but does not warn about similar issues when using literal values or constant expressions.

**To Reproduce**  
Input file (`literal_conversion.sv`):
```systemverilog
module Test;
  logic [15:0] var1;
  logic [31:0] var2;
  logic [63:0] var3;

  initial begin
    // Warning works fine
    var2 = var1;
    var2 = var3;
    var2 = {var1, 1'b0};
    var2 = {var3, 1'b0};

    // Warning not working
    var2 = 64'b0;
    var2 = {32'b0, 32'b0};
    var2 = {$bits(var1){1'b0}};
  end
endmodule
```

Command:

```bash
slang -Wconversion literal_conversion.sv
```

Output:

```
Top level design units:
    Test

literal_conversion.sv:8:12: warning: implicit conversion expands from 16 to 32 bits [-Wwidth-expand]  
    var2 = var1;  
         ~ ^~~~  
literal_conversion.sv:9:12: warning: implicit conversion truncates from 64 to 32 bits [-Wwidth-trunc]  
    var2 = var3;  
         ~ ^~~~  
literal_conversion.sv:10:12: warning: implicit conversion expands from 17 to 32 bits [-Wwidth-expand]  
    var2 = {var1, 1'b0};  
         ~ ^~~~~~~~~~~~  
literal_conversion.sv:11:12: warning: implicit conversion truncates from 65 to 32 bits [-Wwidth-trunc]  
    var2 = {var3, 1'b0};  
         ~ ^~~~~~~~~~~~  

Build succeeded: 0 errors, 4 warnings
```

**Expected behavior**  
All assignments should produce warnings under `-Wconversion` if they involve truncation or expansion, including those using literals (e.g., `64'b0`, `{32'b0, 32'b0}`, or replicated patterns like `{$bits(var1){1'b0}}`).

**Additional context**  
- Slang version: 8.0.142+98534e59  
- Platform: WSL Ubuntu 22.04  
- Compiler: Clang 19  
- Reproducible: Always

MikePopoloski, 2025-04-01:
This is by design to avoid overly annoying warnings; slang will not warn about conversions if the constant value does not truncate any bits. Otherwise even code as simple as this would cause a warning:
```
logic a = 0;
```

hankhsu1996, 2025-04-01:
Understand, thanks!

closed 2025-04-01 by hankhsu1996