MikePopoloski/slang #1296: Visit all SyntaxTree tree

ibar2711, 2025-04-02:
**Is your feature request related to the C++ slang project or the pyslang Python bindings?**
Pyslang Python bindings

**Describe the solution you'd like**
Can you provide an example of traversing the entire tree and getting the data of all nodes from Python?

**Describe alternatives you've considered**
Tried to use visit function but haven't been able to figure out how to traverse the entire tree with this function.

**Additional context**
Add any other context or screenshots about the feature request here.

MikePopoloski, 2025-04-02:
There are examples of visiting syntax trees here: https://github.com/MikePopoloski/slang/blob/master/pyslang/tests/test_visitors.py

ibar2711, 2025-04-02:
Thanks for your response, the quoted example helped me achieve my goal.
However, I would add something like the following to the examples directory to better explain others how  to do it.

def traverse_tree(tree: pyslang.SyntaxTree):

    data = []

    def handle(obj):
        if isinstance(obj, pyslang.SyntaxNode):
            nodeData = {
                "type": str(obj.kind)[11:],# Remove the SyntaxKind prefix
                "sourceRange": obj.sourceRange,
                "text": obj.__str__(),
            }
            data.append(nodeData)

    tree.root.visit(handle)

    for element in data:
        element["startLine"] = tree.sourceManager.getLineNumber(element["sourceRange"].start)
        element["stopLine"] = tree.sourceManager.getLineNumber(element["sourceRange"].end)

    return data

MikePopoloski, 2025-04-03:
I'm always happy to accept PRs!

closed 2025-04-04 by MikePopoloski