MikePopoloski/slang #1297: Internal variable of function are seen affected multiple time

suzizecat, 2025-04-02:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
C++ slang

Seen on the slang frontend for yosys of the latest oss-cad-suite and on current master: `slang version 8.0.145+fed0eff9`

**Describe the bug**
When using a `function` inside several different processes (`always_comb` at least), slang issue the following kind of error: 

```
mwe.sv:17:14: error: variable 'out' driven by always_comb procedure cannot be written to by any other process
        o2 = my_sum(4,5);
             ^~~~~~~~~~~
mwe.sv:13:14: note: also assigned here
        o1 = my_sum(1,3);
             ^~~~~~~~~~~
mwe.sv:3:5: note: original assignment here
    out = a + b;
    ^~~
```

Which seems to denote that the same `out` variable is assigned both time, while `out` is only defined within the function `my_sum` as seen below. 

**To Reproduce**
Using the following snippet: 
```verilog
// mwe.sv
function logic [8:0] my_sum(logic [7:0] a, logic [7:0] b);
    logic [8:0] out;
    out = a + b;
    return out;
endfunction;

module mwe ();

    logic [8:0] o1;
    logic [8:0] o2;

    always_comb begin
        o1 = my_sum(1,3);
    end

    always_comb begin
        o2 = my_sum(4,5);
    end

endmodule
```

use `slang mwe.sv` .

MikePopoloski, 2025-04-02:
I believe the error is correct. `my_sum` is a static function, so there is only one storage location for the variable `out`. This means that on every timestep you are assigning both 4 and 9 to the same variable.

The applicable rule from the LRM 9.2.2.2:
> The variables assigned on the left-hand side of assignments shall not be assigned by any other
process. This includes variables assigned within functions called by the procedure but not those
assigned within tasks called by the procedure.

suzizecat, 2025-04-02:
I found the passage you are mentioning.

So, while this seems very, very counter intuitive to me because, in my software mind, a function is made to be reused (which this statement prevent) and should provide a scope, I wont argue further on this point.

However, it looks like this construct is accepted by all big three tools without any problem (Cadence, Synopsys and Siemens/Questa) in simulation, synthesis and even Lint.
As far as I know, those function would be considered like a non-static functions.

As per LRM 13.4.2 : 
> Functions defined within a module, interface, program, or package default to being static, with all declared
items being statically allocated. These items shall be shared across all uses of the function executing
concurrently.
Functions can be defined to use automatic storage in the following two ways:
> - Explicitly declared using the optional automatic keyword as part of the function declaration.
> - Implicitly declared by defining the function within a module, interface, program, or package that is
defined as automatic.

And indeed, declaring `function automatic logic [8:0] my_sum(logic [7:0] a, logic [7:0] b);` makes it work. (Works also on my actual testcase) 

Would you consider having a flag that default a function declaration as an automatic function ?

suzizecat, 2025-04-04:
For reference, https://github.com/povik/yosys-slang implemented such flag: 
```
--allow-multi-driven-locals      Allow subroutine local variables to be driven from multiple always_comb/_ff blocks
```

povik, 2025-04-04:
> For reference, https://github.com/povik/yosys-slang implemented such flag:
> 
> ```
> --allow-multi-driven-locals      Allow subroutine local variables to be driven from multiple always_comb/_ff blocks
> ```

I should note this flag is provided by slang itself

suzizecat, 2025-04-04:
I completely missed it when checking slang help message (which I did multiple times...)
@MikePopoloski therefore you may close the issue, as the requested flag actually already exists as far as I am concerned. 

Sorry for the noise.

MikePopoloski, 2025-04-04:
Oh yeah, I forgot I had added that already. Hurray.

closed 2025-04-04 by MikePopoloski