MikePopoloski/slang #1299: [slang][serializer] Ast producing bug

likeamahoney, 2025-04-04:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
C++ slang

**Describe the bug**
Related to https://github.com/MikePopoloski/slang/issues/1285. 
So the bug can be reproduced using code sample from this issue using slang `gcc-debug-shared` build.

```verilog
interface bus();
	logic w;
endinterface

module m1(bus intf);
	assign intf.w = 0;
endmodule

module top();
	bus intf1();
	bus intf2();

	m1 a(.intf(intf1));
	m1 b(.intf(intf2));
endmodule
```

```
slang 1.sv --ast-json=1.json
```

`slang` fails with this:

```
internal compiler error: Assertion '!isFrozen()' failed
  in file /home/yan/projects/github/slang/source/ast/Compilation.cpp, line 1673
  function: void slang::ast::Compilation::addDiagnostics(const slang::Diagnostics&)
```

After elaboration the allocator become frozen. So it fails while trying to elaborate cached node bodies during AST serializing.

17Reset, 2025-04-11:
Same issue

Slang Version: 0.8
```
internal compiler error: Assertion '!isFrozen()' failed
  in file D:\Job\Github\slang\source\ast\Compilation.cpp, line 1751
  function: const class slang::ast::Type &__cdecl slang::ast::Compilation::getType(const struct slang::syntax::DataTypeSyntax &,const class slang::ast::ASTContext &,const class slang::ast::Type *)
```

```
internal compiler error: Assertion '!isFrozen()' failed
  in file D:\Job\Github\slang\include\slang/ast/Compilation.h, line 719
  function: class slang::ConstantValue *__cdecl slang::ast::Compilation::allocConstant(class slang::ConstantValue &&)
```

17Reset, 2025-04-11:
I found out that the error might be caused by the attribute annotation, here is the result of the run:

```verilog
(* dont_touch = "yes" *)
module rptr_empty#()();

endmodule : rptr_empty
```
```shell
D:\Download\async_fifo>..\slang08.exe  --ast-json ast.json --top rptr_empty rptr_empty.sv
Top level design units:
    rptr_empty

Build succeeded: 0 errors, 0 warnings
internal compiler error: Assertion '!isFrozen()' failed
  in file D:\Job\Github\slang\include\slang/ast/Compilation.h, line 719
  function: class slang::ConstantValue *__cdecl slang::ast::Compilation::allocConstant(class slang::ConstantValue &&)
```

```verilog
// (* dont_touch = "yes" *)
module rptr_empty#()();

endmodule : rptr_empty
```
```shell
D:\Download\async_fifo>..\slang08.exe  --ast-json ast.json --top rptr_empty rptr_empty.sv
Top level design units:
    rptr_empty

Build succeeded: 0 errors, 0 warnings
```

MikePopoloski, 2025-04-12:
Fixed in 4f7338bac83b2a7376f634f5611e2271db5a7e60

closed 2025-04-12 by MikePopoloski

17Reset, 2025-04-16:
The latest version of slang causes the same problem when dealing with attribute annotation