MikePopoloski/slang #1303: Threading hangs?

jcurtiss8086, 2025-04-06:
Starting with eff44cca85fce53d854816a231fb68b4fad32afc I'm seeing runtime hangs for some repos, but not all.  Running with a debug-build doesn't provide any assertion failures.   

After letting Slang run for a couple of minutes in gdb, here's the backtrace:
```gdb
[New Thread 0x7ffff79e6700 (LWP 14665)]
[New Thread 0x7ffff71e5700 (LWP 14666)]
[New Thread 0x7ffff69e4700 (LWP 14667)]
[New Thread 0x7ffff61e3700 (LWP 14668)]
[New Thread 0x7ffff59e2700 (LWP 14669)]
[New Thread 0x7ffff51e1700 (LWP 14670)]
[New Thread 0x7ffff49e0700 (LWP 14671)]
[New Thread 0x7ffff41df700 (LWP 14672)]
[Thread 0x7ffff41df700 (LWP 14672) exited]
[Thread 0x7ffff49e0700 (LWP 14671) exited]
[Thread 0x7ffff51e1700 (LWP 14670) exited]
[Thread 0x7ffff59e2700 (LWP 14669) exited]
[Thread 0x7ffff61e3700 (LWP 14668) exited]
[Thread 0x7ffff69e4700 (LWP 14667) exited]
[Thread 0x7ffff71e5700 (LWP 14666) exited]
[Thread 0x7ffff79e6700 (LWP 14665) exited]
Top level design units:
    top

[New Thread 0x7ffff79e6700 (LWP 14841)]
[New Thread 0x7ffff71e5700 (LWP 14842)]
[New Thread 0x7ffff69e4700 (LWP 14843)]
[New Thread 0x7ffff61e3700 (LWP 14844)]
[New Thread 0x7ffff59e2700 (LWP 14845)]
[New Thread 0x7ffff51e1700 (LWP 14846)]
[New Thread 0x7ffff49e0700 (LWP 14847)]
[New Thread 0x7ffff41df700 (LWP 14848)]
^C
Thread 1 "slang" received signal SIGINT, Interrupt.
0x00007ffff7f9870c in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
(gdb) bt
#0  0x00007ffff7f9870c in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
#1  0x0000000000437ecd in std::condition_variable::wait<BS::thread_pool<(unsigned char)0>::wait()::{lambda()#1}>(std::unique_lock<std::mutex>&, BS::thread_pool<(unsigned char)0>::wait()::{lambda()#1}) (this=0x7fffffff9300, __lock=..., __p=...) at gcc/12.2.0/include/c++/12.2.0/condition_variable:102
#2  0x0000000000431bf0 in BS::thread_pool<(unsigned char)0>::wait (this=0x7fffffff9278)
    at slang/external/BS_thread_pool.hpp:1798
#3  0x000000000085bc45 in slang::analysis::AnalysisManager::wait (this=0x7fffffff7170)
    at slang/source/analysis/AnalysisManager.cpp:169
#4  0x000000000085a1af in slang::analysis::AnalysisManager::analyze (this=0x7fffffff7170, compilation=...)
    at slang/source/analysis/AnalysisManager.cpp:67
#5  0x000000000041bc39 in slang::driver::Driver::runAnalysis (this=0x7fffffff9750, compilation=...)
    at slang/source/driver/Driver.cpp:917
#6  0x000000000040a86d in driverMain<char**> (argc=203, argv=0x7fffffffb2d8) at slang/tools/driver/slang_main.cpp:185
#7  0x00000000004083bb in main (argc=203, argv=0x7fffffffb2d8) at slang/tools/driver/slang_main.cpp:222
(gdb)

```

jcurtiss8086, 2025-04-07:
Tracked it down to a simple reproducer

```systemverilog
module top;

    function logic[5:0] calc_pagesize(logic [63:12] address, logic s);
        if (s) begin
            casex (address[63:12])
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx0: return 6'd1;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx01: return 6'd2;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx011: return 6'd3;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx0111: return 6'd4;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx01111: return 6'd5;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx011111: return 6'd6;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx0111111: return 6'd7;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx01111111: return 6'd8;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx011111111: return 6'd9;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx0111111111: return 6'd10;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx01111111111: return 6'd11;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx011111111111: return 6'd12;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx0111111111111: return 6'd13;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx01111111111111: return 6'd14;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx011111111111111: return 6'd15;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx0111111111111111: return 6'd16;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx01111111111111111: return 6'd17;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx011111111111111111: return 6'd18;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx0111111111111111111: return 6'd19;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx01111111111111111111: return 6'd20;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx011111111111111111111: return 6'd21;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx0111111111111111111111: return 6'd22;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxx01111111111111111111111: return 6'd23;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxxx011111111111111111111111: return 6'd24;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxxx0111111111111111111111111: return 6'd25;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxxx01111111111111111111111111: return 6'd26;
                52'bxxxxxxxxxxxxxxxxxxxxxxxxx011111111111111111111111111: return 6'd27;
                52'bxxxxxxxxxxxxxxxxxxxxxxxx0111111111111111111111111111: return 6'd28;
                52'bxxxxxxxxxxxxxxxxxxxxxxx01111111111111111111111111111: return 6'd29;
                52'bxxxxxxxxxxxxxxxxxxxxxx011111111111111111111111111111: return 6'd30;
                52'bxxxxxxxxxxxxxxxxxxxxx0111111111111111111111111111111: return 6'd31;
                52'bxxxxxxxxxxxxxxxxxxxx01111111111111111111111111111111: return 6'd32;
                52'bxxxxxxxxxxxxxxxxxxx011111111111111111111111111111111: return 6'd33;
                52'bxxxxxxxxxxxxxxxxxx0111111111111111111111111111111111: return 6'd34;
                52'bxxxxxxxxxxxxxxxxx01111111111111111111111111111111111: return 6'd35;
                52'bxxxxxxxxxxxxxxxx011111111111111111111111111111111111: return 6'd36;
                52'bxxxxxxxxxxxxxxx0111111111111111111111111111111111111: return 6'd37;
                52'bxxxxxxxxxxxxxx01111111111111111111111111111111111111: return 6'd38;
                52'bxxxxxxxxxxxxx011111111111111111111111111111111111111: return 6'd39;
                52'bxxxxxxxxxxxx0111111111111111111111111111111111111111: return 6'd40;
                52'bxxxxxxxxxxx01111111111111111111111111111111111111111: return 6'd41;
                52'bxxxxxxxxxx011111111111111111111111111111111111111111: return 6'd42;
                52'bxxxxxxxxx0111111111111111111111111111111111111111111: return 6'd43;
                52'bxxxxxxxx01111111111111111111111111111111111111111111: return 6'd44;
                52'bxxxxxxx011111111111111111111111111111111111111111111: return 6'd45;
                52'bxxxxxx0111111111111111111111111111111111111111111111: return 6'd46;
                52'bxxxxx01111111111111111111111111111111111111111111111: return 6'd47;
                52'bxxxx011111111111111111111111111111111111111111111111: return 6'd48;
                52'bxxx0111111111111111111111111111111111111111111111111: return 6'd49;
                52'bxx01111111111111111111111111111111111111111111111111: return 6'd50;
                52'bx011111111111111111111111111111111111111111111111111: return 6'd51;
                52'b0111111111111111111111111111111111111111111111111111: return 6'd52;
                default:                                                  return 6'd53;
            endcase
        end else begin
            return 6'd0;
        end
    endfunction

endmodule
```

MikePopoloski, 2025-04-07:
Yeah, that makes sense. The logic for detecting whether the default case will run or not isn't handling the huge bit widths very well. Unfortunately it's an NP-complete problem in the general case so I'll probably need to add a cutoff for large bitwidths.

Also for future debugging help, if you suspect threading you can always run with `-j1` to disable threads.

likeamahoney, 2025-04-11:
Yeah I have the same issue

Canno't elaborate a large designs with new `slang`. For example a little sample from `openc910`:

```verilog
module m();
    reg [63:0] lq_entry_vld, lq_create_ptr1, lq_create_ptr0, misc_ff1_src, misc_ff1_rslt;
    initial begin
  casez(misc_ff1_src[63:0])
    64'b1???????????????????????????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd0;
    64'b01??????????????????????????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd1;
    64'b001?????????????????????????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd2;
    64'b0001????????????????????????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd3;
    64'b00001???????????????????????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd4;
    64'b000001??????????????????????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd5;
    64'b0000001?????????????????????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd6;
    64'b00000001????????????????????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd7;
    64'b000000001???????????????????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd8;
    64'b0000000001??????????????????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd9;
    64'b00000000001?????????????????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd10;
    64'b000000000001????????????????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd11;
    64'b0000000000001???????????????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd12;
    64'b00000000000001??????????????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd13;
    64'b000000000000001?????????????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd14;
    64'b0000000000000001????????????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd15;
    64'b00000000000000001???????????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd16;
    64'b000000000000000001??????????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd17;
    64'b0000000000000000001?????????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd18;
    64'b00000000000000000001????????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd19;
    64'b000000000000000000001???????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd20;
    64'b0000000000000000000001??????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd21;
    64'b00000000000000000000001?????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd22;
    64'b000000000000000000000001????????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd23;
    64'b0000000000000000000000001???????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd24;
    64'b00000000000000000000000001??????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd25;
    64'b000000000000000000000000001?????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd26;
    64'b0000000000000000000000000001????????????????????????????????????: misc_ff1_rslt[63:0] = 64'd27;
    64'b00000000000000000000000000001???????????????????????????????????: misc_ff1_rslt[63:0] = 64'd28;
    64'b000000000000000000000000000001??????????????????????????????????: misc_ff1_rslt[63:0] = 64'd29;
    64'b0000000000000000000000000000001?????????????????????????????????: misc_ff1_rslt[63:0] = 64'd30;
    64'b00000000000000000000000000000001????????????????????????????????: misc_ff1_rslt[63:0] = 64'd31;
    64'b000000000000000000000000000000001???????????????????????????????: misc_ff1_rslt[63:0] = 64'd32;
    64'b0000000000000000000000000000000001??????????????????????????????: misc_ff1_rslt[63:0] = 64'd33;
    64'b00000000000000000000000000000000001?????????????????????????????: misc_ff1_rslt[63:0] = 64'd34;
    64'b000000000000000000000000000000000001????????????????????????????: misc_ff1_rslt[63:0] = 64'd35;
    64'b0000000000000000000000000000000000001???????????????????????????: misc_ff1_rslt[63:0] = 64'd36;
    64'b00000000000000000000000000000000000001??????????????????????????: misc_ff1_rslt[63:0] = 64'd37;
    64'b000000000000000000000000000000000000001?????????????????????????: misc_ff1_rslt[63:0] = 64'd38;
    64'b0000000000000000000000000000000000000001????????????????????????: misc_ff1_rslt[63:0] = 64'd39;
    64'b00000000000000000000000000000000000000001???????????????????????: misc_ff1_rslt[63:0] = 64'd40;
    64'b000000000000000000000000000000000000000001??????????????????????: misc_ff1_rslt[63:0] = 64'd41;
    64'b0000000000000000000000000000000000000000001?????????????????????: misc_ff1_rslt[63:0] = 64'd42;
    64'b00000000000000000000000000000000000000000001????????????????????: misc_ff1_rslt[63:0] = 64'd43;
    64'b000000000000000000000000000000000000000000001???????????????????: misc_ff1_rslt[63:0] = 64'd44;
    64'b0000000000000000000000000000000000000000000001??????????????????: misc_ff1_rslt[63:0] = 64'd45;
    64'b00000000000000000000000000000000000000000000001?????????????????: misc_ff1_rslt[63:0] = 64'd46;
    64'b000000000000000000000000000000000000000000000001????????????????: misc_ff1_rslt[63:0] = 64'd47;
    64'b0000000000000000000000000000000000000000000000001???????????????: misc_ff1_rslt[63:0] = 64'd48;
    64'b00000000000000000000000000000000000000000000000001??????????????: misc_ff1_rslt[63:0] = 64'd49;
[... cut ...]
    64'b000000000000000000000000000000000000000000000000000000000000001?: misc_ff1_rslt[63:0] = 64'd62;
    64'b0000000000000000000000000000000000000000000000000000000000000001: misc_ff1_rslt[63:0] = 64'd63;
    64'b0000000000000000000000000000000000000000000000000000000000000000: misc_ff1_rslt[63:0] = 64'd64;
    default:                                                              misc_ff1_rslt[63:0] = {64{1'bx}};
    end
endmodule
```

@MikePopoloski  Hi, Do you fixing this? Or can I manage this issue?

MikePopoloski, 2025-04-11:
Yes, I will fix it. If it's causing you problems now you can just check out an older commit that still works for you.

MikePopoloski, 2025-04-15:
Should be fixed now

closed 2025-04-15 by MikePopoloski