MikePopoloski/slang #1315: Inconsistent instance array indices

povik, 2025-04-08:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
C++ slang

**Describe the bug**

I don't think slang follows the LRM rule for connecting an unpacked array to an instance array.

From LRM 1800-2017, 23.3.3.5 Unpacked array ports and arrays of instances

> If they match exactly in size, each element of the port connection shall be matched to the port left index to left index, right index to right index.

but as far I can tell that's not what is followed when a little endian unpacked array is connected to a big endian instance array.

**To Reproduce**
Read the following
```
module subm(input h);
endmodule

module top();
	wire g [1:0];
	subm inst[2:3](.h(g));
endmodule
```

`g[0]` is connected to `inst[2].h` but I'd have expected `inst[3].h`.

**Additional context**
The code below in generation of hierarchical names looks weird to me:
https://github.com/MikePopoloski/slang/blob/6e7ad80e6b7b65541097b75a58c99cdfcc38d552/source/ast/Symbol.cpp#L175-L181
first we translate the index to a relative index, and then we add the lower end of the range back in?

Strangely enough the generation of hierarchical names and the resolution of hierarchical references seem to agree on the instace array indices, but the connection of unpacked arrays on a port shows a discrepancy.

MikePopoloski, 2025-04-12:
Indeed, this was pretty confusing. I cleaned it up a bit and fixed the underlying issue in beb5f4ad9d0a4fc460e95bb20c0fd5c6add9ee8d

povik, 2025-04-12:
Oh great, this clears it up for me downstream too.

povik, 2025-04-12:
I believe interface array ports are now getting connected wrong:
```
interface bus(input clk);
	logic a;
endinterface

module m1(bus intf [0:1][2:3]);
	wire w = intf[0][3].a;
endmodule

module top(input logic clk);
	bus top_bus[1:0][1:0](clk);
	m1 m1i(intf);
endmodule
```

The wire `w` should be connected to `top_bus[1][0].a` but it's getting resolved as connected to `top_bus[0][1].a`.

MikePopoloski, 2025-04-15:
Fixed in 23d38707290528b725adf6e15652a90f4d00fb80

closed 2025-04-15 by MikePopoloski