MikePopoloski/slang #1319: Rewriter should fail if attempting to insert invalid syntax nodes

parker-research, 2025-04-12:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
Both (so implementation will be in C++). I will describe the situation in Python here though, as we'll eventually want the fix to have a test case in Python too.

**Describe the bug**
Consider the following test.

```python

def test_rewriter_nested():
    input_tree = pyslang.SyntaxTree.fromText(
        """
        module m;
            struct {
                int int_member_to_be_removed;
                logic logic_member_to_stay_untouched;
            } s;
        endmodule
    """,
        "test.sv",
    )
    expected = pyslang.SyntaxTree.fromText(
        """
        module m;
            struct {
                logic logic_member_to_insert;
                logic logic_member_to_stay_untouched;
            } s;
        endmodule
    """,
        "test.sv",
    )

    check_func_called = {
        "called": False,
        "remove_match_count": 0,
        "insert_match_count": 0,
    }

    def modify_struct(
        node: pyslang.SyntaxNode, rewriter: pyslang.SyntaxRewriter
    ) -> None:
        """Modify the struct with several operations.

        Multiple operations:
            1. Remove `int int_member_to_be_removed` from the struct.
            2. Add `logic logic_member_to_insert` within the struct (at the start).
        """
        check_func_called["called"] = True

        assert isinstance(node, pyslang.SyntaxNode)
        assert isinstance(rewriter, pyslang.SyntaxRewriter)

        # Handle removing `int i;` from inside struct.
        for subnode in node:
            if subnode.kind == pyslang.SyntaxKind.IntType:
                check_func_called["remove_match_count"] += 1
                rewriter.remove(node)
                break

        # Handle adding `logic j` before struct.
        if node.kind == pyslang.SyntaxKind.StructUnionMember:
            print(
                f"StructUnionMember #{check_func_called['insert_match_count']} found: {node}"
            )

            for subnode in node:
                if subnode.kind == pyslang.SyntaxKind.SeparatedList:
                    print(
                        f"SeparatedList #{check_func_called['insert_match_count']} found: {list(subnode)}"
                    )

                    token = subnode[0][0]
                    assert isinstance(token, pyslang.Token)
                    assert token.kind == pyslang.TokenKind.Identifier

                    if token.value == "logic_member_to_stay_untouched":
                        new_decl = pyslang.SyntaxTree.fromText(
                            "logic logic_member_to_insert;", "new.sv"
                        ).root

                        check_func_called["insert_match_count"] += 1

                        rewriter.insert_before(node, new_decl) # THIS SHOULD FAIL

    result = pyslang.rewrite(input_tree, modify_struct)
    assert result is not None
    assert check_func_called["called"] is True, "Handler should have been called"
    assert (
        check_func_called["remove_match_count"] == 1
    ), "Handler should have removed one match"
    assert (
        check_func_called["insert_match_count"] >= 1
    ), "Handler should have inserted one match"
```

**To Reproduce**
Run the code above. It "succeeds", instead of raising in the location where it says "THIS SHOULD FAIL".

The new struct member being inserted is a `DataDeclaration`, but struct types expect their members to be `StructUnionMember` syntax nodes. They happen to stringify to the same thing here but it would not work properly if you wanted to build an AST from this.

**Additional context**
One possible implementation, albeit a somewhat-inefficient one, would be to validate that the round-trip of the resulting AST to a string and then back to an AST are equal. If they are not equal, that indicates that an invalid node was inserted somewhere. Is it true that each Verilog representation only has a single correct AST representation? 

Maybe having a `validate: bool = True` argument added to `.rewrite` might be the best approach, in case anyone ever wants to create intentionally-invalid ASTs.

MikePopoloski, 2025-04-12:
I understand why this would be nice but it's not really possible with the current design. You can't roundtrip from a compiled AST back to the concrete syntax tree(s) that created it, and even if you could, failing to create an AST just means the original code is incorrect, it doesn't tell you that the structure of the concrete syntax tree itself was wrong.

The structure of syntax trees are correct by construction by the Parser class. If you're manually splicing together syntax nodes then you'd need some new validation layer with explicit rules for every kind of syntax node about what is allowed to go where, and a way to keep that validation layer matching the logic in the Parser.

parker-research, 2025-04-12:
Now that I understand better - My new understanding is that the SyntaxNodes are part of the Parse Tree (pre-elaboration). The rewriting is done pre-elaboration.

Why do you suggest that you would need the impossible method of "[converting] from a compiled AST back to the concrete syntax tree(s) that created it"? It seems to me that all this is at the CST/Parse Tree level? 

Allow me to rewrite my propose solution here, and you can tell me if I'm still not getting it - 

> One possible implementation, albeit a somewhat-inefficient one, would be to validate that the round-trip of the resulting _(rewritten)_ ~~AST~~ _CST_ to a string and then back to ~~an AST~~ _a CST_ are equal. If they are not equal, that indicates that an invalid node was inserted somewhere. Is it true that each Verilog representation only has a single correct ~~AST representation~~ _CST representation_?

MikePopoloski, 2025-04-12:
I see, yes, if you serialize your syntax tree to Verilog text, reparse that back to a syntax tree, and then check for equivalence, that probably catches most problems. It is pretty inefficient as you mention but it's not hard to implement.

parker-research, 2025-04-12:
Do you want me to implement this in C++ first and then just do the bindings? Or should I do it in Python only?

MikePopoloski, 2025-04-12:
Yeah, probably should be in C++ and then exposed to Python.