MikePopoloski/slang #1320: Pyslang: Add a `SyntaxNode.to_dict()` and/or `.to_json()` method which converts to the JSON-like tree representation that the `slang` CLI tool yields

parker-research, 2025-04-12:
**Is your feature request related to the C++ slang project or the pyslang Python bindings?**
Pyslang Python bindings

**Is your feature request related to a problem? Please describe.**
At any arbitrary point of AST traversal (e.g., visiting, rewriting), it would be helpful to be able to serialize a node to a dict/json to preview/inspect/debug it. 

The json/dict structure that the `slang` tool outputs in general

I've generated [a quick reference sample file](https://github.com/parker-research/slang-demos/blob/1553f2f/ultraembedded_riscv/demo_3_riscv_exec_only.json) of the format I'm roughly picturing. 

**Describe the solution you'd like**
Above.

**Describe alternatives you've considered**
I'm not aware of a way to use the method shown in `pyslang/examples/driver.py` to generate this AST, but it may be possible. 

Debugging in Pyslang tree traversal is rather painful without this feature.

MikePopoloski, 2025-04-12:
I commented about this in another issue but it seems you've confused the elaborated AST with the parser output SyntaxNode tree. There is no code in slang right now to serialize a SyntaxNode to a structure format, but you can add one if you wish.

parker-research, 2025-04-12:
You're right - thanks for helping me understand. 

Would you suggest implementing it in C++ and then binding it to Python, or is Python the best place to implement it?

MikePopoloski, 2025-04-12:
You can probably do it in just a few lines of Python since all objects are basically dicts anyway so probably start there.

17Reset, 2025-04-14:
Same question, how to implement `slang --ast-json-detailed-types --ast-json xxx.json --top xxx` same function in Pyslang

trile127, 2025-04-22:
For c++, perhaps it would be better to use: https://github.com/nlohmann/json which can let you change any struct/(class?) into json and back

```
struct MyStruct {
        std::string name;

        int start_line;
        int end_line;

        // Default constructor required for deserialization
        MyStruct() = default;
        MyStruct(std::string n, int sl, int el)
            : name(n), start_line(sl), end_line(el) {}

        // Convert Module to JSON
        NLOHMANN_DEFINE_TYPE_INTRUSIVE(MyStruct, name, start_line, end_line)
    };
```

MikePopoloski, 2025-05-24:
Going to close this issue since the Python side is easy and there's a separate issue for the C++ side.

closed 2025-05-24 by MikePopoloski

parker-research, 2025-05-26:
I'd suggest that even though it's easy, it's not done; therefore, the issue shouldn't be closed. 

I'm just getting started on it right now :)

If it's easy, do you have the solution handy?