MikePopoloski/slang #1324: Attribute annotations cause ast generation errors

17Reset, 2025-04-16:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
C++ slang

**Describe the bug**
Attribute annotations cause ast generation errors

I found out that the error might be caused by the attribute annotation, here is the result of the run:

```verilog
(* dont_touch = "yes" *)
module rptr_empty#()();

endmodule : rptr_empty
```
```shell
D:\Download\async_fifo>..\slang08.exe  --ast-json ast.json --top rptr_empty rptr_empty.sv
Top level design units:
    rptr_empty

Build succeeded: 0 errors, 0 warnings
internal compiler error: Assertion '!isFrozen()' failed
  in file D:\Job\Github\slang\include\slang/ast/Compilation.h, line 719
  function: class slang::ConstantValue *__cdecl slang::ast::Compilation::allocConstant(class slang::ConstantValue &&)
```

```verilog
// (* dont_touch = "yes" *)
module rptr_empty#()();

endmodule : rptr_empty
```
```shell
D:\Download\async_fifo>..\slang08.exe  --ast-json ast.json --top rptr_empty rptr_empty.sv
Top level design units:
    rptr_empty

Build succeeded: 0 errors, 0 warnings
```

17Reset, 2025-04-16:
> I am unable to reproduce this issue with the regular slang binary. What is slang08.exe?

slang version 8.0.0+c8d1e61

MikePopoloski, 2025-04-16:
Ok, I figured out how to reproduce. Fixed in 3624203fe287f11f35a82842e1097b1ae624fc19

closed 2025-04-16 by MikePopoloski