MikePopoloski/slang #1336: [Help] How to implement `slang.exe --ast-json ast.json --top top_name *.sv` in Pyslang

17Reset, 2025-04-25:
How can I implement ast json file export in pyslang?I've tried some things, but I can't get it to work.

MikePopoloski, 2025-04-26:
What have you tried?

17Reset, 2025-04-27:
```python
# pyslang: A Python library for parsing SystemVerilog
from pyslang import Driver, CommandLineOptions, DiagnosticEngine

SV_CHECK_CMD: str = 'slang'
SV_CHECK_ARG: list = [
    '--ignore-unknown-modules',
    '--ast-json ast.json',
]

def get_ast_file(
    file_list: list[str], slang_args: list[str] | None = None
) -> str | None:
    # Initialize the slang driver and add standard command-line arguments.
    driver = Driver()
    driver.addStandardArgs()

    # Construct the full command to be passed to the slang driver.
    full_cmd = [SV_CHECK_CMD] + SV_CHECK_ARG + (slang_args or []) + file_list
    full_cmd = ' '.join(full_cmd)

    # Parse the command-line arguments using the slang driver.
    if not driver.parseCommandLine(full_cmd, CommandLineOptions()):
        return None

    # Process options and parse all provided source files.
    if not driver.processOptions() or not driver.parseAllSources():
        return None

    # Create a compilation object and retrieve all diagnostics.
    compilation = driver.createCompilation()
    diags = compilation.getAllDiagnostics()
    source_manager = compilation.sourceManager

    # Generate and return the diagnostic report.
    return DiagnosticEngine.reportAll(source_manager, diags)

if __name__ == '__main__':
    # Example usage of the xqt_rtl_check function.
    file_list = [
        'D:/Download/ProgramFile/rtl_file/async_fifo_old/async_fifo.v',
        'D:/Download/ProgramFile/rtl_file/async_fifo_old/fifomem.v',
        'D:/Download/ProgramFile/rtl_file/async_fifo_old/rptr_empty.v',
        'D:/Download/ProgramFile/rtl_file/async_fifo_old/sync_r2w.v',
        'D:/Download/ProgramFile/rtl_file/async_fifo_old/sync_w2r.v',
        'D:/Download/ProgramFile/rtl_file/async_fifo_old/wptr_full.v',
    ]
    slang_args = ['--top=async_fifo']
    report = get_ast_file(file_list, slang_args)
    if report:
        print(report)
    else:
        print('Error occurred during RTL check.')
```

output:
```
slang: unknown command line argument '--ast-json', did you mean '--diag-json'?
```

17Reset, 2025-04-27:
I was just wondering how to implement the following command in pyslang: `slang.exe --ast-json ast.json --top top_name *.sv`.Due to the scarcity of pyslang documentation, it's hard to find a way to do this

MikePopoloski, 2025-04-27:
I see, yes the C++ version of AST serialization is not exposed to Python. Python makes it so easy to serialize arbitrary objects to JSON (or whatever other format) that it was assumed you would just write something yourself to make the format how you want it.

17Reset, 2025-04-28:
Does this mean that I can format an exact C++ version of the AST syntax tree through Python's serialization capabilities?

17Reset, 2025-04-28:
Thank you very much, with C++ source and Python binding I managed to use `pyslang` to restore ast.json in the same mode as slang cli. However, this process ran into what seems to be a `bug`:
The `DefinitionSymbol` class defined in `Symbolbindings.cpp` is shown below, but the `cellDefine` property is not found in the corresponding `pyslang`.

```cpp
    py::class_<DefinitionSymbol, Symbol>(m, "DefinitionSymbol")
        .def_readonly("definitionKind", &DefinitionSymbol::definitionKind)
        .def_readonly("defaultLifetime", &DefinitionSymbol::defaultLifetime)
        .def_readonly("unconnectedDrive", &DefinitionSymbol::unconnectedDrive)
        .def_readonly("cellDefine", &DefinitionSymbol::cellDefine)
        .def_readonly("timeScale", &DefinitionSymbol::timeScale)
        .def_property_readonly("defaultNetType",
                               [](const DefinitionSymbol& self) { return &self.defaultNetType; })
        .def_property_readonly("instanceCount", &DefinitionSymbol::getInstanceCount)
        .def("getKindString", &DefinitionSymbol::getKindString)
        .def("getArticleKindString", &DefinitionSymbol::getArticleKindString)
        .def("__repr__", [](const DefinitionSymbol& self) {
            return fmt::format("DefinitionSymbol(\"{}\")", self.name);
        });
```

pyslang __init__.pyi
```python
class DefinitionSymbol(Symbol):
    @staticmethod
    def _pybind11_conduit_v1_(*args, **kwargs):
        ...
    def __repr__(self) -> str:
        ...
    def getArticleKindString(self) -> str:
        ...
    def getKindString(self) -> str:
        ...
    @property
    def defaultLifetime(self) -> VariableLifetime:
        ...
    @property
    def defaultNetType(self) -> ...:
        ...
    @property
    def definitionKind(self) -> DefinitionKind:
        ...
    @property
    def instanceCount(self) -> int:
        ...
    @property
    def timeScale(self) -> TimeScale | None:
        ...
    @property
    def unconnectedDrive(self) -> UnconnectedDrive:
```

[image]

[image]

MikePopoloski, 2025-04-28:
I'm sorry, I'm having a hard time understanding what you're trying to say. You can see in the binding code that the cellDefine member is exposed.

17Reset, 2025-04-29:
Although the binding code shows that `cellDefine` has been exposed to Python, in the actual `pyslang.pyd`, there is no such member function, and the call will result in an error.
```
Traceback (most recent call last):
  File "D:\xxx\slang_demo.py", line 301, in <module>
    dump_ast_to_json(source_files, output_file=f'{output_path}/async_fifo_ast.json')
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\xxx\slang_demo.py", line 20, in dump_ast_to_json
    "definitions": [serialize_symbol(definition) for definition in compilation.getDefinitions()]
                    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "D:\xxx\slang_demo.py", line 58, in serialize_symbol
    result["cellDefine"] = symbol.cellDefine
                           ^^^^^^^^^^^^^^^^^
AttributeError: 'pyslang.DefinitionSymbol' object has no attribute 'cellDefine'
```

MikePopoloski, 2025-04-29:
Have you compiled and installed the Python extension?

17Reset, 2025-04-30:
Of course

```
D:\Download>python -V
Python 3.13.3

D:\Download>pip show pyslang
Name: pyslang
Version: 8.0.0
Summary: Python bindings for slang, a library for compiling SystemVerilog
Home-page: https://sv-lang.com/
Author: Mike Popoloski
Author-email:
License: Copyright (c) 2015-2025 Michael Popoloski

Permission is hereby granted, free of charge, to any person obtaining a copy
of this software and associated documentation files (the "Software"), to deal
in the Software without restriction, including without limitation the rights
to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the Software is
furnished to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in
all copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
THE SOFTWARE.
Location: D:\App\Python\Lib\site-packages
Requires:
Required-by:
```

17Reset, 2025-05-06:
I found the reason, it was a version issue, the version of pyslang that exists on pypi is not up to date, the property is available in the latest version. Hopefully it's possible to update pyslang on pypi.💐

closed 2025-05-06 by MikePopoloski