MikePopoloski/slang #1343: Can you easily extract the Include directive

Fromhell777, 2025-04-29:
**Is your feature request related to the C++ slang project or the pyslang Python bindings?**
C++ slang

**Is your feature request related to a problem? Please describe.**
I would like to extract the Include directive after/during creating the syntax tree. Right now this is my current approach:
* Create the syntax tree. This runs the pre-processor under the hood 
* I gave a SourceManager to the syntax tree which is used by this pre-processor
* Here SourceManger opens all the included files (also recursively) and they are available in a vector of buffers
* I can check whether a buffer in the SourceManager was included in another file with the getIncludedFrom(buffer) function
* I can then check if this SourceLocation is valid with the isIncludedFileLoc(sourceLocation) function
* Then I can request the std::filesystem::path with the getFullPath(buffer) function
* But I can't seem to get access to the original include directive. Mainly if the file contains "`include "bla.v"" I don't see how I can extract the "bla.v" part. I seems that for Macro's you have something in place to get the original macro with the getMacroNam() function. Would it be possible to do the same for Include directive names?
* The getFileName() and getRawFileName() functions only give the relative path to the included files. This does not seem to be enough info to do the task

**Describe the solution you'd like**
I would like to easily extract the filename of the inlcude directives. This can either be from the SourceManager or some place else. The SourceManager seems the most logical at this point

**Describe alternatives you've considered**
I could try to do my own string processing with the SourceLocations that I can extract from the SourceManager

Fromhell777, 2025-04-30:
I also quickly tried to extract the MacroName or just the original Macro text before it was expanded but I wasn't able to make it work

MikePopoloski, 2025-05-06:
Hmm yeah, this is similar to https://github.com/MikePopoloski/slang/issues/815 in that an API for what you want doesn't exist, and it's currently not straightforward because any given filename can actually map to multiple buffers since it could be included from multiple locations (not to mention macro locations that come from those buffers).

AndrewNolte, 2025-05-22:
SyntaxTree::getIncludeDirectives() was added in https://github.com/MikePopoloski/slang/pull/1355, which I think is what you are looking for here.