MikePopoloski/slang #1344: Fails to parse implicit parameter

pointerliu, 2025-04-30:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
C++ slang

**Describe the bug**
Slang fails to accept named parameter assignment in a `bind` statement when only the parameter name is provided without an explicit value, such as `.MAX_COUNT`. Instead, it currently only accepts the `.MAX_COUNT(MAX_COUNT)` form.

**To Reproduce**
The following code should parse successfully, but instead results in a syntax error:

```systemverilog
module dut #(parameter int MAX_COUNT = 9) ();
endmodule

module checker #(parameter int MAX_COUNT = 9) ();
endmodule

module bind_wrapper;
  bind dut checker #(
    .MAX_COUNT  // This line causes a parse error, but is legal SystemVerilog
  ) inst_checker ();
endmodule
```

```bash
top.sv:4:15: error: expected '('
    .MAX_COUNT
              ^
```

MikePopoloski, 2025-04-30:
I do not believe this is legal SystemVerilog. Furthermore I could not find a single other tool that allows it even as an extension.

jcurtiss8086, 2025-05-01:
Unfortunately, only port-connections allow .name or .* syntax, but parameters do not.  It's quite cumbersome when dealing with heavily parameterized/reusable designs and binding checkers into the DUT.

pointerliu, 2025-05-02:
I found this usage in project [Ibex](https://github.com/lowRISC/ibex/blob/8a3d46f095db817fafdad9de4d157ccb83a267fb/dv/verilator/simple_system_cosim/ibex_simple_system_cosim_checker_bind.sv):

```systemverilog
module ibex_simple_system_cosim_checker_bind;
  bind ibex_simple_system ibex_simple_system_cosim_checker#(
      .SecureIbex,
      .ICache,
      .PMPEnable,
      .PMPGranularity,
      .PMPNumRegions,
      .MHPMCounterNum
  ) u_ibex_simple_system_cosim_checker_bind (
      .clk_i            (IO_CLK),
      .rst_ni           (IO_RST_N),
      .host_dmem_req    (host_req[CoreD]),
      .host_dmem_gnt    (host_gnt[CoreD]),
      .host_dmem_we     (host_we[CoreD]),
      .host_dmem_addr   (host_addr[CoreD]),
      .host_dmem_be     (host_be[CoreD]),
      .host_dmem_wdata  (host_wdata[CoreD]),
      .host_dmem_rvalid (host_rvalid[CoreD]),
      .host_dmem_rdata  (host_rdata[CoreD]),
      .host_dmem_err    (host_err[CoreD])
  );
endmodule
```
This seems to be accepted by Verilator. Maybe I'm misunderstanding something...

MikePopoloski, 2025-05-02:
Hmm, I guess it's a Verilator extension then. Or more likely a bug / oversight.