MikePopoloski/slang #1353: add import package parser

ddppt-yy, 2025-05-13:
**Is your feature request related to the C++ slang project or the pyslang Python bindings?**
Pyslang Python bindings

**Is your feature request related to a problem? Please describe.**
return header.imports is a string

**Describe the solution you'd like**
Perhaps the names of its packages and  import parameters can be parsed
eg.
input：
```verilog
import xx_pkg::*;  
import
 oo_pkg::AAA
     ;
```
output
header.imports.members, its a dict, key is package name, and value is parameter:
{
    'xxx_pkg' : '*',
    'oo_pkg' : 'AAA'
}

MikePopoloski, 2025-05-13:
What have you tried?

MikePopoloski, 2025-05-18:
Closing since there's been no followup.

closed 2025-05-18 by MikePopoloski

ddppt-yy, 2025-05-19:
sorry， I want to use pyslang to parse the library files used in the sv file, but I find that the return value is only a string instead of a specific package. So I have this question: Is my method wrong or is detailed parsing not supported at present?

ddppt-yy, 2025-05-23:
```verilog
module

test 

import xx_pkg::*;  
import
 oo_pkg::*
     ;
//aaa
#(
    parameter logic [2-1:0] NN = 3,
    parameter logic [4-1:0] AA = 8,
    parameter logic [8-1:0] BB =16,
    parameter logic [AA+BB-1:0] CC = AA-10
    +BB*123
......
```

```python
tree = pyslang.SyntaxTree.fromFile('test.sv')
mod = tree.root.members[0]

get_type_kind_dir(mod.header)
print(len(mod.header))

print(mod.header.imports)

```
output：
```
import xx_pkg::*;  
import
 oo_pkg::*
     ;

```