MikePopoloski/slang #1354: Macro definitions (`ifdef`, ``define`) deal with the problem

17Reset, 2025-05-13:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
Pyslang Python bindings

**Describe the bug**
Load original code with macro definitions (`ifdef`, ``define`) that are lost after pyslang.

**To Reproduce**
source code:
```SystemVerilog
`define ENABLE_FEATURE_X // Define a macro for conditional compilation

module comment_example #(
    parameter WIDTH = 8 // Parameter for data width, with a line-end comment
) (
    input  logic clk,       
    input  logic rst_n,     
);

    // This is a comment starting at the beginning of a line, before any code.
    logic [WIDTH-1:0] internal_reg; // An internal register

`ifndef DISABLE_DEBUG_MESSAGES
    // This block is active if DISABLE_DEBUG_MESSAGES is NOT defined
    always_comb begin
        // Combinational logic for debugging (if enabled)
        if (data_in > 100) begin
            $display("Debug: data_in (%0d) is greater than 100 at time %0t", data_in, $time); // Line-end comment
        end
    end
`endif

endmodule // End of module comment_example
```

```python
import pyslang

def print_source_code(source: str):
    tree = pyslang.SyntaxTree.fromFile(source)

    syntax_node = tree.root

    for member in syntax_node.members:
        print(str(member))

if __name__ == '__main__':
    file_path = "./format_test.sv"
    print_source_code(file_path)
```

output：
```
 // Define a macro for conditional compilation
module comment_example #(
    parameter WIDTH = 8 // Parameter for data width, with a line-end comment
) (
    input  logic clk,       
    input  logic rst_n,     
);
    // This is a comment starting at the beginning of a line, before any code.
    logic [WIDTH-1:0] internal_reg; // An internal register
    // This block is active if DISABLE_DEBUG_MESSAGES is NOT defined
    always_comb begin
        // Combinational logic for debugging (if enabled)
        if (data_in > 100) begin
            $display("Debug: data_in (%0d) is greater than 100 at time %0t", data_in, $time); // Line-end comment
        end
    end
endmodule
```

MikePopoloski, 2025-05-13:
The skipped tokens are still available as Trivia in the syntax tree and can be printed with the SyntaxPrinter by setting appropriate options.

17Reset, 2025-05-14:
Why are macro definitions, preprocessing, etc. categorized as Trivia

MikePopoloski, 2025-05-14:
Because they can appear anywhere in the parse tree and aren't representable in the grammar.

17Reset, 2025-05-15:
Okay, got it. Thanks.