MikePopoloski/slang #1356: Generic `interconnect` bit width is always 0 with `Type::getBitstreamWidth()`

akashlevy, 2025-05-15:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
C++ slang and `yosys-slang`

**Describe the bug**
There does not seem to be a way to dynamically resolve the width of an `UntypedType`. Calling `Type::getBitstreamWidth()` on an `UntypedType` always returns 0.

To fix this, you would probably need to implement dynamic resolution in a similar fashion to how SV `class` width is implemented. I don't currently see any methods to resolve currently. Curious to hear your thoughts.

**To Reproduce**
See https://github.com/povik/yosys-slang/issues/146

**Additional context**
Context is that the `interconnect` width needs to be resolved for `yosys-slang` to be able to dump into RTLIL.

MikePopoloski, 2025-05-15:
I think the idea is that interconnect nets aren't real and don't ever have an actual width. They don't end up in the final simulation or synthesized output, you have to resolve them to a real net somewhere.

23.3.3.7.1 describes how they get merged (in not very great detail). In particular:
> It shall be illegal for the type of a simulated net (see 23.3.3.7) at the end of elaboration to be an
interconnect net.

I think slang is probably not implementing that check right now. There is logic to do the merging of net types in `PortSymbol::checkSimulatedNetTypes` that could be rearranged into something you can use more generally. Note that every "bit" of the connection can be a different net type!

akashlevy, 2025-05-16:
I was not aware of `PortSymbol::checkSimulatedNetTypes`, thanks for the helpful pointer.

MikePopoloski, 2025-08-26:
I was looking at this some more (because of the linked PR) and I think there's more to do here beyond just checks for interconnect nets, so here are some notes for later. slang implements the dissimilar net type checks listed in the LRM but only the checks; what you really want is a way to query what the actual net type of each net should be, which because of the port collapsing rules means it may not be the net type that was written on the declaration.

For example:
```
module netlist;
  wire w;
  dut1 child1(w);
  dut2 child2(w);
  initial #10 $display(w);
endmodule
module dut1(wand w);
  assign w = 1;
endmodule
module dut2(wor w);
  assign w = 0;
endmodule
```
Wire `w` gets merged as a `wand` net by the connection to `child1`, and then the connection to `child2` should warn because `wor` can't merge with `wand` but right now slang does not warn because it looks at each port in isolation. In order to simulate this correctly you need to know that `w` ends up as a `wand` net, so the display should print `0`. If you reverse the order of `child1` and `child2` you instead get a `wor` net and the display will print `1`. 

Doing this correctly needs a global view of assignments so probably needs to move to the analysis layer. And then when examining any net symbol you need to look up with the analysis results what the actual simulated net type should be (remembering again that each individual bit of a multi-bit net can have a different net type).