MikePopoloski/slang #1369: `error: generate loop hit maximum step limit; possible infinite loop?` on divide by 2

gadfort, 2025-05-23:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
C++ slang

**Describe the bug**
When elaborating a design with a parameter that is divided by 2, slang generates an error, but if the design is modified to right shift the error goes away.

**To Reproduce**
[test.tar.gz](https://github.com/user-attachments/files/20413511/test.tar.gz)
```
tar xvf test.tar.gz
cd test
slang -top cacheLRU cacheLRU.sv
```

There are two files `cacheLRU.sv` and `cacheLRU_working.sv` the first one is the original file that generates an error and second works in slang.
```
127c127
<   for(node = NUMWAYS/2-1; node >= 0; node--) begin
---
>   for(node = (NUMWAYS>>1)-1; node >= 0; node--) begin
```

**Additional context**
Add any other context about the problem here. If you have a non-standard setup or anything unusual that may be contributing to the problem it's good to include that here.

gadfort, 2025-05-23:
@MikePopoloski yes, the current head works. Do you think it would be possible to generate a new release 8.1? since I would need the pyslang updated to handle this?

Thanks

MikePopoloski, 2025-05-24:
v8.1 has been released

closed 2025-05-24 by MikePopoloski

gadfort, 2025-05-27:
@MikePopoloski thank you