MikePopoloski/slang #1380: [bug] Bug with AST traversing due to instance caching

likeamahoney, 2025-05-27:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
C++ slang 

Hi all!

I came across a bug when I used the `slang` api to write a traversal of all members of all instances in DFS order.

Assertion `isFrozen` is invoked  at slang gcc debug build.

You can reproduce it using that patch for `slang_main` driven. Here just a simple DFS with visiting all members recursively after compilation and analyses passes:

```diff
diff --git a/tools/driver/slang_main.cpp b/tools/driver/slang_main.cpp
index f5ba9b877..c162a69ab 100644
--- a/tools/driver/slang_main.cpp
+++ b/tools/driver/slang_main.cpp
@@ -10,6 +10,7 @@
 #include <iostream>
 
 #include "slang/ast/ASTSerializer.h"
+#include "slang/ast/ASTVisitor.h"
 #include "slang/ast/Compilation.h"
 #include "slang/ast/symbols/CompilationUnitSymbols.h"
 #include "slang/diagnostics/TextDiagnosticClient.h"
@@ -227,6 +228,13 @@ int driverMain(int argc, TArgs argv) {
                 TimeTraceScope timeScope("semanticAnalysis"sv, ""sv);
                 driver.runAnalysis(*compilation);
             }
+            for (auto instance : compilation->getRoot().topInstances) {
+                instance->visit(makeVisitor([&](auto& visitor, const InstanceBodySymbol& inst) {
+                    for (auto& member : inst.members()) {
+                        member.visit(visitor);
+                    }
+                }));
+            }
 
             ok &= driver.reportDiagnostics(quiet == true);

```

Then you can run `slang`:

```
slang bug.sv
```

using `SystemVerilog` file below:

```verilog
module top;
    design_top topt();
endmodule

module design_top;
    m2 chk1();
    m2 chk2();
endmodule

module m2;
    m3 chk3();
endmodule

module m3();
endmodule
```

as a result:

```
internal compiler error: Assertion '!isFrozen()' failed
```

@MikePopoloski Is this method of visiting all members of all instances wrong? Am I doing something wrong? Or is this a bug?

MikePopoloski, 2025-05-28:
Yes, you must skip non-canonical instances after the compilation has been frozen. Alternatively you can run your visitor before analysis, or I suppose you could add a method to unfreeze the compilation after analysis is done if you need it run there for some reason.

likeamahoney, 2025-05-28:
> you must skip non-canonical instances after the compilation has been frozen

Manually I can't do this without significant changing in my visitor hierarchy and adding new handlers (such as `InstanceSymbol` handler) to check is body canonical or not.

In provided example above i have only one `InstanceBodySymbol` handler which visiting it's members. And there is no way to check is this canonical or not.

MikePopoloski, 2025-05-28:
You could also just disable instance caching if you want. The point of instance caching is to skip over non-canonical instances, so if you actually want to visit them then disabling it should be fine. Analysis is done multithreaded which is why the Compilation needs to be frozen -- any mutation would be a race condition.

likeamahoney, 2025-05-28:
> You could also just disable instance caching if you want. The point of instance caching is to skip over non-canonical instances, so if you actually want to visit them then disabling it should be [...]

Okay now it's clear for me! Thank you so much! I will close this issue

closed 2025-05-28 by likeamahoney