MikePopoloski/slang #1390: Odd memory issues with pyslang

therealshreyas, 2025-06-02:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
Pyslang Python bindings

**Describe the bug**
I have a script where I parse the tokens from a test.sv (the same exact test.sv from [here](https://sv-lang.com/pyslang/)).
I then call a function which creates a copy of the list of tokens and prints them (this function is called _func_without_dict_). I then call another function (called _func_with_dict_) which creates a useless dictionary, and then calls func_without_dict. 
For some reason, _func_with_dict_ causes a segmentation fault! Using lldb, the segfault is caused by a bad memory access, which I find odd in this case since there is virtually no change in _func_with_dict_

**To Reproduce**
I've created a repository with test.sv and script.py [here](https://github.com/therealshreyas/pyslang_bug). The segmentation fault occurs from calling $python3 script.py

**Additional context**
Please run script.py with a local build of Slang:
git clone https://github.com/MikePopoloski/slang.git
cd slang
pip install .

therealshreyas, 2025-06-02:
I also get the following error -- UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte -- when trying to print str(k).

MikePopoloski, 2025-06-04:
Hmm, I am unable to reproduce this. It's certainly possible there are lurking lifetime issues here though. The SyntaxTree object holds all of the memory for the parsed tree. All of the SyntaxNodes, Tokens, etc are allocated there, so if it gets destroyed you get dangling references. The bindings are supposed to note this relationship to keep the parent alive but it could be wrong in some way. If you change the script to make sure you reference the `tree` later at the end of the test so that Python can't GC it, does the problem go away?

MikePopoloski, 2025-07-11:
Closing since I can't reproduce and there's been no followup.

closed 2025-07-11 by MikePopoloski