MikePopoloski/slang #1397: [slang-netlist] Bug with duplication of the same assignment

likeamahoney, 2025-06-06:
Hi @jameshanlon 

There is an another bug with such code:

```verilog
module aes_key_mem1(input wire  key);
 reg  key_mem_new;
   always_comb
     begin: round_key_gen
             key_mem_new   = key;
             key_mem_new   = key;
     end
endmodule // aes_key_mem
```

There is no combinational loop it is just 2 continuous blocking assignment on the same variable but `slang-netlist` warns on it:

```
Detected 1 combinatorial loop:
Path length: 3
combloopsbug.sv:7:30: note: variable key read from
             key_mem_new   = key;
                             ^~~

combloopsbug.sv:7:14: note: variable key_mem_new assigned to
             key_mem_new   = key;
             ^~~~~~~~~~~
```

If leave only one assignment tool will not found any combinational loop.

likeamahoney, 2025-06-09:
Minimal example from the topic was inspired by real-world example like this:

```verilog
module aes_key_mem(input wire [255 : 0]  key);
 reg [3 : 0] round_ctr_reg;
 reg [127 : 0] key_mem_new;
   always @*
     begin: round_key_gen
  if (round_ctr_reg == 0)
          begin
             key_mem_new   = key[255 : 128];
    end
        else if (round_ctr_reg == 1)
          begin
             key_mem_new   = key[127 : 0];
    end
     end
endmodule // aes_key_mem
```

alexeyvoh, 2025-06-10:
Also wrong loop detection:

````
module test();
   int apb_xx_paddr;
   assign psel_s5 = apb_xx_paddr>=1 && apb_xx_paddr <=6;
endmodule // test
````
````
Detected 1 combinatorial loop:
Path length: 3
log/cl_06.sv:3:40: note: variable apb_xx_paddr read from
   assign psel_s5 = apb_xx_paddr>=1 && apb_xx_paddr <=6;
                                       ^~~~~~~~~~~~

log/cl_06.sv:3:11: note: variable psel_s5 assigned to
   assign psel_s5 = apb_xx_paddr>=1 && apb_xx_paddr <=6;

````