MikePopoloski/slang #1416: pyslang `std::get: wrong index for variant` error on OpenTitan

gussmith23, 2025-06-19:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
Pyslang

**Describe the bug**
Getting a `RuntimeError` when parsing an OpenTitan file.

**To Reproduce**

```python
import pyslang

tree = pyslang.SyntaxTree.fromText("""
// Copyright lowRISC contributors (OpenTitan project).
// Licensed under the Apache License, Version 2.0, see LICENSE for details.
// SPDX-License-Identifier: Apache-2.0
//
// Serial Peripheral Interface (SPI) Device module.
//

`include "prim_assert.sv"

package spi_device_pkg;

  // Passthrough Inter-module signals
  typedef struct packed {
    // passthrough_en: switch the mux for downstream SPI pad to host system not
    // the internal SPI_HOST IP
    logic       passthrough_en;

    // Passthrough includes SCK also. The sck_en is pad out enable not CG
    // enable. The CG is placed in SPI_DEVICE IP.
    logic       sck;
    logic       sck_en;

    // CSb should be pull-up pad. In passthrough mode, CSb is directly connected
    // to the host systems CSb except when SPI_DEVICE decides to drop the
    // command.
    logic       csb;
    logic       csb_en;

    // SPI data from host system to downstream flash device.
    logic [3:0] s;
    logic [3:0] s_en;
  } passthrough_req_t;

  typedef struct packed {
    // SPI data from downstream flash device to host system.
    logic [3:0] s;
  } passthrough_rsp_t;

  parameter passthrough_req_t PASSTHROUGH_REQ_DEFAULT = '{
    passthrough_en: 1'b 0,
    sck:            1'b 0,
    sck_en:         1'b 0,
    csb:            1'b 1,
    csb_en:         1'b 0,
    s:              4'h 0,
    s_en:           4'h 0
  };

  parameter passthrough_rsp_t PASSTHROUGH_RSP_DEFAULT = '{
    s: 4'h 0
  };

  // Command Type
  //
  // Passthrough module does not strictly follow every bit on the SPI line but
  // loosely tracks the phase of the commands.
  //
  // The command in SPI Flash can be categorized as follow:
[... 672 lines cut ...]
Symbol(SymbolKind.FormalArgument, "ci_addr_mode")
Symbol(SymbolKind.FormalArgument, "addr_4b_en")
Symbol(SymbolKind.Variable, "get_addr_mode")
Symbol(SymbolKind.Variable, "result")
Symbol(SymbolKind.TransparentMember, "CmdInfoReadStatus1")
Symbol(SymbolKind.TransparentMember, "CmdInfoReadStatus2")
Symbol(SymbolKind.TransparentMember, "CmdInfoReadStatus3")
Symbol(SymbolKind.TransparentMember, "CmdInfoReadJedecId")
Symbol(SymbolKind.TransparentMember, "CmdInfoReadSfdp")
Symbol(SymbolKind.TransparentMember, "CmdInfoReadCmdStart")
Symbol(SymbolKind.TransparentMember, "CmdInfoReadCmdEnd")
Symbol(SymbolKind.TransparentMember, "CmdInfoReserveStart")
Symbol(SymbolKind.TransparentMember, "CmdInfoReserveEnd")
Symbol(SymbolKind.TransparentMember, "CmdInfoEn4B")
Symbol(SymbolKind.TransparentMember, "CmdInfoEx4B")
Symbol(SymbolKind.TransparentMember, "CmdInfoWrEn")
Symbol(SymbolKind.TransparentMember, "CmdInfoWrDi")
Symbol(SymbolKind.TransparentMember, "NumTotalCmdInfo")
Traceback (most recent call last):
  File "....", line 590, in <module>
    compilation.getRoot().visit(visitor)
  File "....", line 587, in visitor
    print(node)
RuntimeError: std::get: wrong index for variant
```

MikePopoloski, 2025-06-20:
I cannot reproduce. What version of pyslang do you have installed? Is that the full code you are running?

MikePopoloski, 2025-06-26:
Ok well, going to close since there's been no followup.