MikePopoloski/slang #1429: unknown command line argument '+libext'

huaixil, 2025-07-02:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
C++ slang

**Describe the bug**
we have "+libext+.v" in the lib.f file and when we run the command
`slang --top top_wrap -f src/lib.f`
We got the error
```
slang: unknown command line argument '+libext'
slang: unknown command line argument '+libext'
```

Then we tried `slang --top top_wrap --cmd-rename "+libext","--libext" -f src/lib.f`
still
```
slang: unknown command line argument '--libext+.v'
```
And it seems that it can't recognize `+libext+`

**To Reproduce**

**Additional context**

huaixil, 2025-07-03:
Yeah, the .f file is from an existing project, so we can't modify it. All I want is to find a way to convert
`+libext+.v` to `--libext .v` or slang can support `+libext`
Thanks!

MikePopoloski, 2025-07-15:
Fixed in 28fe6a78a065588c4aeba0b86b78253fe76a8a05 and 163fe1f3db99ef0b1ebf28563bca5bdb22eefe2b