MikePopoloski/slang #1439: Does not build on WebAssembly WASI

whitequark, 2025-07-14:
**Is your feature request related to the C++ slang project or the pyslang Python bindings?**
C++ slang

**Is your feature request related to a problem? Please describe.**
I would like slang to be ported to WASI, a POSIX subset for WebAssembly.

To be clear, I plan to port it myself. I'm opening this issue as a way to discuss some of the issues I encounter in the process, as there are bits of Slang internals I don't understand.

whitequark, 2025-07-14:
@MikePopoloski It looks like `SmallMap` is completely broken on 32-bit platforms. I get this error:

```
In file included from /home/whitequark/Projects/yowasp/yosys/yosys-slang-src/third_party/slang/source/ast/symbols/CompilationUnitSymbols.cpp:10:
In file included from /home/whitequark/Projects/yowasp/yosys/yosys-slang-src/third_party/slang/source/ast/symbols/ParameterBuilder.h:11:
/home/whitequark/Projects/yowasp/yosys/yosys-slang-src/third_party/slang/source/../include/slang/util/SmallMap.h:71:23: error: static assertion failed due to requirement '24UL % 16UL == 0': size N needs to be a multiple of alignment Align
   71 |         static_assert(N % Align == 0, "size N needs to be a multiple of alignment Align");
      |                       ^~~~~~~~~~~~~~
/home/whitequark/Projects/yowasp/yosys/yosys-slang-src/third_party/slang/source/../include/slang/util/SmallMap.h:122:27: note: in instantiation of member function 'slang::detail::hashing::StackAllocator<std::pair<const std::string_view, unsigned long>, 24>::StackAllocator' requested here
  122 |     SmallMap() : BaseType(Alloc(*this)) {}
      |                           ^
/home/whitequark/Projects/yowasp/yosys/yosys-slang-src/third_party/slang/source/ast/symbols/CompilationUnitSymbols.cpp:429:43: note: in instantiation of member function 'slang::SmallMap<std::string_view, unsigned long, 2>::SmallMap' requested here
  429 |     SmallMap<std::string_view, size_t, 2> topCellNames;
      |                                           ^
/home/whitequark/Projects/yowasp/yosys/yosys-slang-src/third_party/slang/source/../include/slang/util/SmallMap.h:71:33: note: expression evaluates to '8 == 0'
   71 |         static_assert(N % Align == 0, "size N needs to be a multiple of alignment Align");
      |                       ~~~~~~~~~~^~~~
1 error generated.
```

It's not super clear to me how to fix the allocator, do you have an idea?

MikePopoloski, 2025-07-15:
You can get rid of those static_asserts; they're hidden behind `#if !defined(_M_IX86)` but I suppose that doesn't apply to other 32-bit targets like WASM. They're mostly sanity checking the lack of extra padding bytes on 64-bit targets.

AnalysisManager is not the only user of threads; SourceLoader does so also. Probably you need to add a CMake option SLANG_USE_THREADS to opt out pulling pthread into the build, and then you might as well use that to set a define you can refer to in the code to compile away the threading bits.

Interesting assertion about the alignment; what is `alignof(std::max_align_t)` on WASM? This might be fixable by explicitly providing the alignment template argument of the StackAllocator instead of relying on the default of max_align_t.

whitequark, 2025-07-15:
> You can get rid of those static_asserts; they're hidden behind `#if !defined(_M_IX86)` but I suppose that doesn't apply to other 32-bit targets like WASM. They're mostly sanity checking the lack of extra padding bytes on 64-bit targets.

Got it. Should I add a check for pointer width in CMake and gate them behind that check, switch them to less-than-or-equal, or remove them entirely?

> AnalysisManager is not the only user of threads; SourceLoader does so also. Probably you need to add a CMake option SLANG_USE_THREADS to opt out pulling pthread into the build, and then you might as well use that to set a define you can refer to in the code to compile away the threading bits.

Got it: you are fine with `#ifdef`s (as opposed to other ways one might consider structuring them). I'll add an option.

> Interesting assertion about the alignment; what is alignof(std::max_align_t) on WASM? This might be fixable by explicitly providing the alignment template argument of the StackAllocator instead of relying on the default of max_align_t.

It's 16. Should I instantiate it with Align=16?

whitequark, 2025-07-15:
Actually, would something like this work?

```
template<typename T, size_t N, size_t Align = alignof(T)>
class StackAllocator {
```

MikePopoloski, 2025-07-15:
You can just remove those two static_asserts in Token.h, they aren't needed anymore.

Setting the alignment argument like that should work.

whitequark, 2025-07-16:
> AnalysisManager is not the only user of threads; SourceLoader does so also.

I was wondering why I've never hit compilation errors there and it looks like SourceLoader is only used by the driver. What does the driver even do in Slang? Is it basically a linter?