MikePopoloski/slang #1443: Overly conservative race detection for separate array index assignments

JacyCui, 2025-07-18:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
C++ slang

**Describe the bug**
The slang compiler incorrectly reports a race condition in SystemVerilog code where there is actually no conflict between processes.
The error occurs when one process writes to `data[0]` via `always_comband` another writes to` data[i]` (where `i` starts from 1) via `always_ff`.
These writes are to different elements of the array and should not conflict.

**To Reproduce**

Compile the following SystemVerilog code with slang:

```systemverilog
//--------------------------------------------------------------------------------
// dynamic_delay.sv
// published as part of https://github.com/pConst/basic_verilog
// Konstantin Pavlov, pavlovconst@gmail.com
//--------------------------------------------------------------------------------

// INFO --------------------------------------------------------------------------------
//  Dynamic delay for arbitrary signal.
//
//  Incoming data elements have WIDTH bits each. Module does serialization of
//  input data and outputs flattened bits, based on provided selector value.
//  You can perform delays bit-wize, not just element-wize.
//
//  CAUTION: Be careful selecting last, most-delayed "WIDTH" number of bits.
//           The module intentionally does NOT implement "out of range"
//           checks. Please handle them externally.

/* --- INSTANTIATION TEMPLATE BEGIN ---

dynamic_delay #(
  .LENGTH( 3 ),
  .WIDTH( 4 )
) M (
  .clk( clk ),
  .nrst( nrst ),
  .ena( 1'b1 ),
  .in( in_data[3:0] ),
  .sel( sel[3:0] ),
  .out( out_data[3:0] )
);

--- INSTANTIATION TEMPLATE END ---*/

module dynamic_delay #( parameter
  LENGTH = 63,                        // maximum delay chain length
  WIDTH = 4,                          // data width

  SEL_W = $clog2( (LENGTH+1)*WIDTH )  // output selector width
                                      //  plus one is for zero delay element
)(
  input clk,
  input nrst,
  input ena,
  input [WIDTH-1:0] in,               // input data
                                      // bit in[0] is the "oldest" one
                                      // bit in[WIDTH] is considered the most recent
  input [SEL_W-1:0] sel,              // output selector
  output logic [WIDTH-1:0] out        // output data
);

logic [(LENGTH+1)-1:0][WIDTH-1:0] data = '0;

// packed vector includes extra bits
logic [(LENGTH+1)*WIDTH-1:0] pack_data;
assign pack_data[(LENGTH+1)*WIDTH-1:0] = data;

integer i;
always_ff @(posedge clk) begin
  if( ~nrst ) begin
    // reset all data except zero element
    for( i=1; i<(LENGTH+1); i=i+1 ) begin
      data[i][WIDTH-1:0] <= '0;
    end
  end else if (ena) begin
    for( i=1; i<(LENGTH+1); i=i+1 ) begin
      data[i][WIDTH-1:0] <= data[i-1][WIDTH-1:0];
    end
  end
end

integer j;
always_comb begin
  // zero element assignment
  data[0][WIDTH-1:0] <= in[WIDTH-1:0];

  // output selector, sel==0 gives non-delayed output
  for( j=0; j<WIDTH; j=j+1 ) begin
    out[j] <= pack_data[sel[SEL_W-1:0]+j];
  end
end

endmodule
```

**Expected behavior​​**

The code should compile without errors since there is no actual conflict between the processes:

- `always_ff` only writes to `data[i]` where `i` ranges from `1` to `LENGTH`

- `always_comb` only writes to data[0]

**Additional context**
The error message suggests there's a race condition where none exists.
This appears to be an overly conservative analysis by the compiler that doesn't properly account for the non-overlapping index ranges being accessed by the different processes.

MikePopoloski, 2025-07-19:
This is not legal in SystemVerilog according to the LRM; see https://sv-lang.com/user-manual.html#loop-unroll for more explanation. slang and other tools have an extension to allow it by using heuristics for `for` loops with known iteration bounds, but at least for slang those heuristics are defeated if you use a static variable declared elsewhere as the loop induction variable. You can fix it by declaring `i` inside the for loop. This could potentially be improved for this specific case but it will always be possible to write your code in such a way that the heuristics don't apply.

JacyCui, 2025-07-20:
Thank you for the detailed reply and explanation—it makes a lot of sense now. I really appreciate the quick response.

Verilog and SystemVerilog are indeed very complex languages, and it's clear that there's often a noticeable gap between the IEEE specification and how things are implemented or used in practice. I can see how tools like slang have to strike a careful balance, and I appreciate the effort that goes into handling these subtleties.

Thanks again for the great work on slang!